PerMalmberg/Smooth #4: Undefined refernce to esp pthread set cfg

enelson1001, 2018-11-09:
I am trying to build a simple hello world. I am using VSCode for IDE and have been using nkolban esp32 snippets without any problem.  This is the error I am seeing on the build.

LD build/hello_world.elf
/home/ed/esp32/projects/sandbox/hello_world/build/smooth/libsmooth.a(Task.o):(.literal._ZN6smooth4core4Task5startEv+0x10): undefined reference to `esp_pthread_set_cfg(esp_pthread_cfg_t const*)'
/home/ed/esp32/projects/sandbox/hello_world/build/smooth/libsmooth.a(Task.o): In function `smooth::core::Task::start()':
/home/ed/esp32/esp-idf/components/smooth/include/smooth/core/Task.h:78: undefined reference to `esp_pthread_set_cfg(esp_pthread_cfg_t const*)'
collect2: error: ld returned 1 exit status
/home/ed/esp32/esp-idf/make/project.mk:405: recipe for target '/home/ed/esp32/projects/sandbox/hello_world/build/hello_world.elf' failed
make: *** [/home/ed/esp32/projects/sandbox/hello_world/build/hello_world.elf] Error 1
The terminal process terminated with exit code: 2

Any idea what is going on?

I am using commit  2f32356. as I was having problems with the latest commit.  Even with this commit I had to change smooth/include/smooth/core/json/Value.h line #6 from #include <json/cJSON/cJSON.h> to #include <cJSON.h> or else compiler could not find thecJSON.h file.

Thanks

PerMalmberg, 2018-11-09:
What version of ESP-IDF are you using? I'm developing against the latest master.

enelson1001, 2018-11-09:
Resolved - switched to latest (master branch) of esp-idf and changed IDE to
Eclipse.

Still had to change smooth/include/smooth/core/json/Value.h line #6 from
#include <json/cJSON/cJSON.h> to #include <cJSON.h> or else compiler could
not find the cJSON.h file.

Thanks

enelson1001, 2018-11-11:
> That is odd, somehow you have an include path to
> ${IDF_PATH}/components/json/cJSON instead of ${IDF_PATH}/components.
>
> Did you regenerate the project?

Per
I am using Eclipse Photon
To get project to compile without errors and to have the #includes resolved
correctly I had to do the following.
1. To get the #includes to resolve correctly:
     a. Projects-> properties -> C/C++ General -> Paths and Symbols -> Go
to Includes tab then click add button and add ${IDF_PATH}/components to GNU
C and GNU C++ "Include directories".
      b. If I open smooth/include/smooth/core/json/Value.h and go to
#include <json/cJSON/cJSON> and right click and select Open Declaration I
will see the cJSON.h file.

2. To get The project to compile without the fatal error:
json/cJSON/cJSON.h: No such file or directory I had to do the following.
     a. Modify component.mk file located in esp-idf/components/json so it
looked like the following.
#
# Component Makefile
#
COMPONENT_ADD_INCLUDEDIRS := ../          *<-- THIS IS NEW*
#COMPONENT_ADD_INCLUDEDIRS := cJSON       *<--- THIS IS ORIG*
COMPONENT_SRCDIRS := cJSON
COMPONENT_SUBMODULES := cJSON
COMPONENT_OBJS := cJSON/cJSON.o cJSON/cJSON_Utils.o

Then in the build directory: build/json the component_project_vars.mk
looked like this.
# Automatically generated build file. Do not edit.
COMPONENT_INCLUDES += $(IDF_PATH)/components  *<-- THIS IS WITH MOD*
COMPONENT_LDFLAGS += -L$(BUILD_DIR_BASE)/json -ljson
COMPONENT_LINKER_DEPS +=
COMPONENT_SUBMODULES += $(IDF_PATH)/components/json/cJSON
COMPONENT_LIBRARIES += json
component-json-build:

Before the modification line #1 looked like this
COMPONENT_INCLUDES += $(IDF_PATH)/components/json/cJSON *<--ORIG*

Not sure if this is the correct way to fix the problem but it worked?????

Thanks

PerMalmberg, 2018-11-11:
Ah, so you're not using it as a CMake-project; that explains why you have to make changes. I use CLion ad my IDE and while IDF's CMake-ish [setup isn't yet really finished](https://www.esp32.com/viewtopic.php?f=13&t=7535) it works well enough until Espressif release a new version of the CMake support.

enelson1001, 2018-11-13:
Per
I am trying to go thru the test code you placed under the test folder.  The
2nd test program I am trying to run is under "queue" but App::event(const
timer::ElapsedTime& event) never gets called.  If I add log.info to
SenderTask::tick() I do see log.info that tic is being called.
1. The terminal displays (without the log.info inside tick())the following.
I (2706828) SocketDispatcher: Minimum free stack: 4700 of 8192 => 3492 used.
I (2706838) SenderTask: Minimum free stack: 7156 of 8192 => 1036 used.
2. Is something missing in sdcard.cpp or queue.h?
3. My main.cpp looks like this.
#include <queue.h>
extern "C" {
    void app_main()
    {
        queue::App app;
        app.start();
    }
}
4. If I change SenderTask tick interval from 1 millisecond to 500
milliseconds I see the following message on terminal.
E (10424) task_wdt: Task watchdog got triggered. The following tasks did
not reset the watchdog in time:
E (10424) task_wdt:  - IDLE1 (CPU 1)
E (10424) task_wdt: Tasks currently running:
E (10424) task_wdt: CPU 0: main
E (10424) task_wdt: CPU 1: pthread
5. Why does QueueNotification.h have #include "QueueNotification.h" inside
the file?

Thanks

PerMalmberg, 2018-11-13:
1. Looks like the main task is starved of CPU cycles. Are you running with the sdconfig that comes with smooth or a custom one?
2. No? What does sdcard.cpp have to do with the queue example?
3. As expected.
4. The idle task easily gets starved of CPU cycles and unless you actually utilize it to perform work, it is nothing to worry about.
5. That's a mistake Fixed in [this commit](https://github.com/PerMalmberg/Smooth/commit/da2ac2d9869f861e1c0fbd2db67a459c5dda9ffb) 

I'll try to remember to run the queue example when I get back home to double check that it works.

PerMalmberg, 2018-11-13:
@enelson1001 I just ran the queue example without modifications (using the sdkonfig file included with the Smooth test project) and it behaves as expected with the following output:

> (V)Queue: Creating queue 'string queue-TaskEventQueue', with 10 items of size 32.
(D)MainTask: Running as attached thread
(D)MainTask: Executing...
(V)MainTask: Initializing...
(V)Queue: Creating queue 'SocketDispatcher-TaskEventQueue', with 10 items of size 8.
(V)Queue: Creating queue 'SocketOperations-TaskEventQueue', with 20 items of size 12.
(D)SocketDispatcher: Creating worker thread
(D)SocketDispatcher: Waiting for worker to start
(D)SocketDispatcher: Executing...
(D)SocketDispatcher: Notify start_mutex
(D)SenderTask: Creating worker thread
(D)SenderTask: Waiting for worker to start
(V)SocketDispatcher: Initializing...
(V)SocketDispatcher: Initialized
(D)SenderTask: Executing...
(D)SenderTask: Notify start_mutex
(V)SenderTask: Initializing...
(V)SenderTask: Initialized
(V)MainTask: Initialized
(D)Stats: Avg: 226.03us per message, Count: 100, Last msg: 130us
(D)Stats: Avg: 312.69us per message, Count: 200, Last msg: 129us
(D)Stats: Avg: 290.99us per message, Count: 300, Last msg: 129us
(D)Stats: Avg: 310.39us per message, Count: 400, Last msg: 129us
(D)Stats: Avg: 279.65us per message, Count: 500, Last msg: 130us
(D)Stats: Avg: 288.78us per message, Count: 600, Last msg: 130us
(D)Stats: Avg: 288.39us per message, Count: 700, Last msg: 129us
(D)Stats: Avg: 272.25us per message, Count: 800, Last msg: 130us
(D)Stats: Avg: 308.07us per message, Count: 900, Last msg: 1000us
(D)Stats: Avg: 297.42us per message, Count: 1000, Last msg: 129us
(D)Stats: Avg: 288.01us per message, Count: 1100, Last msg: 129us
(D)Stats: Avg: 289.48us per message, Count: 1200, Last msg: 129us
(D)Stats: Avg: 283.30us per message, Count: 1300, Last msg: 129us
(D)Stats: Avg: 280.42us per message, Count: 1400, Last msg: 129us
(D)Stats: Avg: 275.51us per message, Count: 1500, Last msg: 130us
(D)Stats: Avg: 269.08us per message, Count: 1600, Last msg: 129us
(D)Stats: Avg: 267.06us per message, Count: 1700, Last msg: 129us

PerMalmberg, 2018-11-14:
Great. I'm guessing it is the FreeRTOS tick that was different. I think I set it to the maximum 1000 to be able to get the highest possible timer resolution.

closed 2018-11-14 by PerMalmberg