PerMalmberg/Smooth #5: Are these includes needed?

enelson1001, 2018-11-14:
Per
First off I am no where near your experience as a C++ programmer, I still am a novice.
I was looking thru the code to try an understand how Task Event Queue works (still trying to figure it out) and saw some includes that I had questions on.  There may be reason why you include stuff that I am not aware of (maybe future enhancements?) but here is what i have found so far.  

**TaskEventQueue.h**
//
// Created by permal on 6/27/17.
//

#pragma once

#include <smooth/core/Task.h>
#include "ITaskEventQueue.h"
#include "IEventListener.h"
#include "Link.h"                                 //// <----- is this required
#include "QueueNotification.h"
#include <smooth/core/logging/log.h>

**QueueNotification.h**
//
// Created by permal on 10/18/17.
//

#pragma once

#include <chrono> 
#include <queue>
#include <mutex>
#include <algorithm>
#include <condition_variable>    ////////////////  <--- added ?????
#include "ITaskEventQueue.h"
#include <smooth/core/timer/ElapsedTime.h>  ////////// <---- is this required

namespace smooth
{
    namespace core
    {
        //class Task;  ////////////////// <---------Is this required

**QueueNotifications.cpp**
//
// Created by permal on 10/18/17.
//

#include <thread>                                  /// <----- is this required ?
#include <smooth/core/Task.h>                      /// <------ is this required ?
#include <smooth/core/ipc/QueueNotification.h>
#include <smooth/core/timer/ElapsedTime.h>         //// <----- is this required ?
#include <smooth/core/logging/log.h>

 Thanks

PerMalmberg, 2018-11-15:
For a novice, you have a keen eye. :) Some of those might not be needed, and are likely leftovers from earlier iterations. I'll have a look at cleaning the includes.

PerMalmberg, 2018-11-15:
Some where needed, some where not. Also added an example for the ISRTaskEventQueue.

commit 64cbf76f (2018-11-15): Removed some unneeded includes.
Bonus: Added example project for ISRTaskEventQueue.
Closes #5
closed 2018-11-15 by PerMalmberg