PerMalmberg/Smooth #7: Main task and Btask not running

enelson1001, 2018-11-22:
Per
I modified your hello world program and adding a second task and broke out the tasks into separate files.
This is my project structure.
GreetingCPPv2
     |
     ------- CMakeList.txt
     ------- partitions.csv
     -------- sdkconfig
     ------- build
     ------- include
                 |
                  ------- GreetingCPPv2.h
                  -------- ATask.h
                  -------- BTask.h
     ------- main
                 |
                 ------ CMakeList.txt
                 ------ GreetingCPPv2.cpp
                 ------- ATask.cpp
                 ------- BTask.cpp
                 ------ main.cpp
I can compile the program without any errors but this is the output I see. As you can see the Main Task is not running and the BTask is not running.  It looks like MainTask and BTask don't finish initializing?
[image]

This is my top level CMakeList.txt
/home/ed/vscode-workspace-esp32/sandbox/GreetingCPPv2/CMakeLists.txt

This is GreetingCPPv2.h
/home/ed/vscode-workspace-esp32/sandbox/GreetingCPPv2/include/GreetingCPPv2.h

This is ATask.h
/home/ed/vscode-workspace-esp32/sandbox/GreetingCPPv2/include/ATask.h

This is BTask.h
/home/ed/vscode-workspace-esp32/sandbox/GreetingCPPv2/include/BTask.h

This is GreetingCPPv2.cpp
/home/ed/vscode-workspace-esp32/sandbox/GreetingCPPv2/main/GreetingCPPv2.cpp

This is ATask.cpp
/home/ed/vscode-workspace-esp32/sandbox/GreetingCPPv2/main/ATask.cpp

This is BTask.cpp
/home/ed/vscode-workspace-esp32/sandbox/GreetingCPPv2/main/BTask.cpp

This is main.cpp
/home/ed/vscode-workspace-esp32/sandbox/GreetingCPPv2/main/main.cpp

This is CMakeList.txt inside main directory
/home/ed/vscode-workspace-esp32/sandbox/GreetingCPPv2/main/CMakeLists.txt

Can you tell me what I did wrong?
Thanks

PerMalmberg, 2018-11-22:
No I cannot because you're not showing any code. (Please either link or use gists instead of pasting the code into the issue itself)

enelson1001, 2018-11-22:
Here is zip file of files
[GreetingCPPv2.zip](https://github.com/PerMalmberg/Smooth/files/2608610/GreetingCPPv2.zip)

Also did Gist but not sure how you see the files

PerMalmberg, 2018-11-23:
I've not run your code, but try setting FreeRTOS tick rate to 1000Hz (you have it at 100Hz) and let me know if that help. I'm quite sure that's the problem.

enelson1001, 2018-11-23:
Per
Changed to 1000 but same problem.
1. I added the following to main.ccp to verify tick rate.
       #include <freertos/FreeRTOSConfig.h>
       std::cout << "Tick rate = " << configTICK_RATE_HZ << std::endl;
and it printed out 1000.

2. To get the program to link without any errors I had to comment out the
ESP_PLATFORM check line 134 and line 166 in file
esp-idf/components/smooth/CMakeLists.txt or else I was getting the
following:
    undefined reference to `smooth::core::network::Wifi::connect_to_ap()'
    undefined reference to `smooth::core::network::Wifi::Wifi()'
3. If I leave the check as is add a message inside the ESP_PLATFORM check
it prints my message that it is a ESP_PLATFORM but still get the linker
errors as stated in #2
4. I am using Visual Studio Code + C/C++ IntelliSense + CMakeTools + CMake
version 3.12.4
5. Something strange is going on?

Thanks

PerMalmberg, 2018-11-24:
First, you'd need to create a public gist and link to it. Or better, simply link to your repository so it can be cloned; that way anyone trying to help you would get everything exactly the way you have set it up.

Secondly, now that you seem to have moved over to using CMake as the build system you need to know that I'm running on a custom branch of ESP-IDF containing changes to get the _ESP_PLATFORM_ which you're missing. My fork is [located here](https://github.com/PerMalmberg/esp-idf). I've submitted a [PR](https://github.com/espressif/esp-idf/pull/2601) to get this into the upstream master, but no response from Espressif yet.

Thirdly, I recommend you to use the same project layout as the Smooth examples, so:

* Add smooth as as a submodule or at least clone it to the externals/smooth folder:
`git submodule add https://github.com/PerMalmberg/Smooth.git externals/smooth`
or
`git clone https://github.com/PerMalmberg/Smooth.git externals/smooth`

* In your top level CMakeLists.txt you have to add the following line:
`set(EXTRA_COMPONENT_DIRS ${CMAKE_CURRENT_LIST_DIR}/externals/smooth/lib)`

* In your main/CMakeLists.txt add this line:
`set(COMPONENT_REQUIRES smooth)`

That will make the ESP_IDF build system properly include smooth and also make your app link against the smooth library. (Assuming you use my ESP-IDF fork linked above).

Making only the above changes to your files, I flashed the app to my ESP32 Wrover module and I get the expected result. I even left `CONFIG_FREERTOS_HZ=100` as-is, though you need to change that too 1000 to get proper timing for anything below a second.

From a `build` directory inside the project directory:
```
~/esp/esp-idf/tools/idf.py -C .. -p /dev/ttyUSB1 reconfigure
~/esp/esp-idf/tools/idf.py -C .. -p /dev/ttyUSB1 app-flash monitor
```

[image]

enelson1001, 2018-11-25:
Per
First I created a github repository for the Smooth testing located here
https://github.com/enelson1001/GreetingCPP. See Getting-Started.txt under
vscode-workspace-esp32.

I tried your suggestions starting with the HelloCPP project:
1. Used your fork for the esp-idf.
2. Cloned you Smooth to create an externals/smooth library under HelloCPP
project directory
3. Modified the two CMakeLists.txt files and added your suggestions.

But
1. CMake would fail with the following errors.
[cmake] CMake Error at
/home/ed/esp32-smooth/esp-idf/tools/cmake/kconfig.cmake:122
(add_custom_target):
[cmake]   add_custom_target cannot create target "menuconfig" because
another target
[cmake]   with the same name already exists.  The existing target is a
custom target
[cmake]   created in source directory
[cmake]   "/home/ed/vscode-workspace-esp32/sandbox/HelloCPP".  See
documentation for
[cmake]   policy CMP0002 for more details.
[cmake] Call Stack (most recent call first):
[cmake]   /home/ed/esp32-smooth/esp-idf/tools/cmake/project.cmake:107
(kconfig_process_config)
[cmake]   externals/smooth/lib/smooth/CMakeLists.txt:196 (project)
[cmake]
[cmake]
[cmake] CMake Error at externals/smooth/lib/smooth/CMakeLists.txt:202
(target_include_directories):
[cmake]   Cannot specify include directories for target "smooth" which is
not built
[cmake]   by this project.

The only way to get rid of the errors was to comment out the ESP_PLATFORM
checks in lib/smooth/CMakeLists.txt.  See the Readme.txt file under
externals/smooth directory for the changes I made to the file.  After I
made the changes the CMake errors went away and I could compile and build
without any problems.  I was able to flash the device and everything worked

Then I tried the GreetingCPP project.  The project that started this issue
and you got to work on your end.
1. Copied the modified externals/smooth directory from HelloCPP to
GreetingCPP project so it was using the modified lib/smooth/CMakeLists.txt
file.
2. No CMake errors but get now I get build errors.
[build] [513/809  62% :: 1543119781.868] Building CXX object
smooth/CMakeFiles/smooth.dir/core/timer/ElapsedTime.cpp.obj
[build] [514/809  62% :: 1543119781.868] Building CXX object
smooth/CMakeFiles/smooth.dir/core/Application.cpp.obj
[build] [515/809  62% :: 1543119781.869] Building CXX object
smooth/CMakeFiles/smooth.dir/application/network/mqtt/Subscription.cpp.obj
[build] [516/809  62% :: 1543119781.873] Building CXX object
smooth/CMakeFiles/smooth.dir/core/json/Value.cpp.obj
:
:
:
main/CMakeFiles/main.dir/main.cpp.obj.d -o
main/CMakeFiles/main.dir/main.cpp.obj -c ../main/main.cpp
[build] In file included from ../main/main.cpp:7:0:
[build] ../include/GreetingCPPv2.h:3:38: fatal error:
smooth/core/Appliccation.h: No such file or directory
[build] compilation terminated.

Your are probably getting tired of this so I would understand if you did
not want to provide any further assistance.

Thanks

PerMalmberg, 2018-11-25:
If you still had to comment out the ESP_PLATFORM checks, then you're likely not actually using my fork of ESP-IDF.

I suggest you carefully study how the test projects are set up in the smooth framework, but also [this other project](https://github.com/PerMalmberg/IO-Card-G3/tree/master/software) which uses the framework.

enelson1001, 2018-11-25:
Per
Of course I used your fork.

I have asked for assistance from CMake-Tools to help solve the problem.  I
think it is because you are using CLion and I am using VSCode. When I put
debug statements in cmake.project and your /smooth/lib/smooth/CMakeList.txt
files I noticed your CMakeLists.txt file is called 3 times, 2 times it
knows that it is an ESP_PLATFORM and 1 time it does not.
I'll let you know what I find out.

If your curious I have added a folder to my repository called
CMake-Problems.

Thanks

PerMalmberg, 2018-11-26:
I'm using CLion, but still build via the python tools provided by IDF (CLion can't generate the build scripts due to issues with the xtensa-gcc). Espressif are about to merge an update to the CMAke build system so a lot of this will likely be changed. Hopefully that will result in us being able to use CMake as it is intended.

enelson1001, 2018-11-26:
Per
I got it working - happy days.

I did not receive reply from CMake Tools so I did some further debugging.

It looks like line 166 - add_subdirectory(${COMPONENT_PATH}
${COMPONENT_NAME}) makes two call to lib/smooth/CMakeLists.txt where the
first call ESP_PLATFORM is not set and the 2nd call ESP_PLATFORM is set.

To get it working I ended up doing was the following.
In your fork of the esp-idf I modified esp-idf/tools/project.cmake file.
1. Added set(ESP_PLATFORM "1") after line 6 set(IDF_PATH "$ENV{IDF_PATH}")
2. Changed line 81 from -D "ESP_PLATFORM=1" to -D
"ESP_PLATFORM=${ESP_PLATFORM}"

Not sure why using a variable was better instead of "1" but with the above
changes I did not receive any cmake errors.

Then reviewing your CMakeList.txt files from IO-Card-G3 I was able to
modify my CMakeLists.txt files and everything works.

I would be nice if your readme page would say something that if people want
to use cmake they should use your fork of esp-idf.

Now I can get back to my project.

One last question - how many tasks can you run on the ESP32?

You can close this issue.

Thanks

PerMalmberg, 2018-11-27:
How many tasks you can run depends on how much memory you have free and how much each one uses.

I'll update the readme with some info.

closed 2018-11-27 by PerMalmberg