PerMalmberg/Smooth #17: Platform IO

fab-san, 2019-04-26:
Hi,

I looked at your library and I like it quite a bit.
Have you considered register Smooth with Platform IO? It would be very easy and useful.

PerMalmberg, 2019-04-26:
Hi @fab-san 

Glad to hear you like it. I'd love to hear what you've done with it and if you have any thoughts on how it can be improved.

Since I decided that PlatformIO wasn't for me a year or two back, I haven't looked at it again and I honestly don't know what you mean by "registering Smooth with PIO". Do they have some registry for libraries or such?

fab-san, 2019-04-26:
Hi there!

I am still learning it but I will surely let you know what I find. I am working on an interface between my home automation system and the HVAC. I designed and built a I/O board around a EST32 with a touch screen. I have my own little library, but now that I found Smooth, I would love to port everything to it (I haven't started yet).

Yes, they have a register of third-party libraries. One can write a JSON file with the download information of the library and a URL to the zip file (Github is the recommended way). Once the library is registered, it is publicly available for anyone using Platform IO. Smooth can be downloaded, built, and used directly through it.

I am Fabrizio by the way.

PerMalmberg, 2019-04-26:
Hello Fabrizio :)

At this time I've got my hands full of getting my own project finished (currently writing the Secure server sockets and HTTP(s) server), so its not something I want to spend time on. If you are using PlatformIO and want to use Smooth in it, feel welcome to do a PR with the required changes. That I will probably have time to look at.

fab-san, 2019-04-29:
Absolutely!

closed 2019-05-02 by PerMalmberg