PerMalmberg/Smooth #49: Support for websocket?

squonk11, 2019-07-12:
Hi, congratulations for these nice classes for the ESP32. I am currently searching for a HTTP(S) webserver software with the following attributes:
1. esp-idf based (not Arduino)
2. c++ based
3. TLS cabable
4. support for websockets (according to RFC6455)
I see that 1. - 3. is already provided by your classes. The only thing missing for me is the support for websocket communication. Is there a plan to implement the websocket protocol also?
Best regards
   Lothar

PerMalmberg, 2019-07-12:
@squonk11 I'm actually currently thinking about how to best implement minimal, but functional, websocket support. If you have suggestions on how, feel free to contribute. PRs are accepted.

squonk11, 2019-07-13:
Unfortunately I am not a professional SW engineer but a privately interested enthusiast. I implemented already the websocket protocol into [this](https://github.com/fhessel/esp32_https_server) repository. But since this is Arduino I am asking myself if it is easier to port this to esp-idf or implement websocket into "smooth". 
I had a short look into your sources but until now I did not get the point how things work together. I see that you are using some quite sophisticated features of C++ which I am currently not so familiar with (as I said: I am not a professional SW engineer). Somewhere there must be the analysis of the received header - here the detection of the websocket connection must be implemented. Then the websocket connection must not be closed after the response - here it is also important to understand how this can be implemented.
Maybe you have some explanations which makes it easier for me to understand your sourecode?

PerMalmberg, 2019-07-13:
> some quite sophisticated features of C++

Not really, mainly template and polymorphism. I'm sure you can understand it given time and a bit of help :)

The main classes of interest are likely `application::network::http::HTTPProtocol` and `application::network::http::HTTPServerClient`. As everything is templated all the way down to the `core::network::Socket` class, I'm thinking its probably best to extend the `HTTPProtocol` such that it can operate in two modes - regular HTTP verbs and Websocket. This can probably be done using either inheritance or composition; I'm thinking that the later is the better option since Websockets really aren't the same protocol has regular HTTP.

[`HTTPProtocol::data_received`](https://github.com/PerMalmberg/Smooth/blob/master/lib/smooth/application/network/http/HTTPProtocol.cpp#L49) is likely the place to handle eventual upgrade-requests and once detected, switch "modes" and start delivering data in a format suitable for websockets to the applications. Currently [`HTTPPacket`](https://github.com/PerMalmberg/Smooth/blob/master/lib/smooth/include/smooth/application/network/http/HTTPPacket.h) is used for this, so either that has to be changed, or some other way has to be found. An important part of the Smooth framework is that everything is event based so this will have to be too, meaning that just as there may be several `HTTPPacket` received by the application (e.g. a file is [POST:ed by the client](https://github.com/PerMalmberg/Smooth/blob/master/test/http_server_test/http_server_test.cpp#L264)) in a single requests, websockets may have to split data into several parts too (unless Websockets guarantee a max packet size?)

Keep-alive is already property of the `HTTPSServerClient` so that functionality is easily achieved.

I'll be thinking about this a bit more while doing some gardening. Btw, I'm usually on IRC in #ESP32 @ freenode, as permal if you want to chat.

PerMalmberg, 2019-07-13:
So I gave this some thought. I think the proper way is to rework the `HTTPProtocol` class such that it holds either a "regular HTTP protocol" or a "websocket protocol" implementation, but only one at a time. This way we can keep memory usage to a minimum, while still allowing an upgrade transition at runtime. It also plays nicely with the templatization.

PerMalmberg, 2019-07-13:
@squonk11 As you can see, I've begun an implementation of websockets, currently working on the connection upgrade. Do you happen to have a good debugging tool for it, Firefox doesn't give any info on why it doesn't accept the websocket connection from /websocket.html, [generated here](https://github.com/PerMalmberg/Smooth/blob/feature/49-websocket/lib/smooth/include/smooth/application/network/http/HTTPServer.h#L414)?

squonk11, 2019-07-14:
Sorry for not answering yesterday - Saturdays and/or Sundays I have to take care for my father. 
Thank you for already taking care of this topic. I think today I will have some time to take a look at the work you have done so far.

PerMalmberg, 2019-07-14:
No worries, @squonk11. IRL is important :)

squonk11, 2019-07-14:
Sorry, but I did not get much time to check your sourcecode. What I have seen so far is that you already provided the skeleton for websockets and the detection of the upgrade. But reading the RegularHTTPProtocol.cpp (in order to make a similar WebsocketProtocol.cpp) is still a bit difficult for me as I am not so familiar with templates. I think I need to read first a little bit about using templates.
Additionally I am doing such work during my free time (which currently is quite small) - I would like to support/co-work with you on this but it will take time...

PerMalmberg, 2019-07-14:
@squonk11 Its OK. I wanted this functionality myself so don't feel you owe me anything. I'm currently working out how to decode the incoming frames and how to re-use/transform the `HTTPPacket` class to act as a carrier for both regular HTTP and Websocket data.

If you have any gotchas from when you implemented the protocol feel free to share them.

FYI: I'm not going to support data frames with sizes given by 64 bits, it just doesn't make sense for an embedded device.

squonk11, 2019-07-14:
When I implemented the websocket server I implemented a WebsocketHandler class with virtual functions for onMessage, onClose and onError which the end-user has to override. For reading the data I was using a std::streambuf. 
I encountered two pitfalls during the implementation:

1. The extended payload length in the websocket frame  is a 16 bit (or 64 bit) value in network byte order; so, that ntohs() (or ntohl() ) have to be used. 
2. The received data has to be de-masked only if the mask bit in the websocket frame is set.

Your approach not to implement frame sizes with 64bit size probably makes sense - although in  my application I have to transmit a file with a size of approx. 170kB. Here 64k (16bit) is too small. In order to handle that amount of data I will be using a Wroover module with 8MB PSRAM. But I think I can also cut the file into pieces of <64kB.

PerMalmberg, 2019-07-15:
I realized that handling 65k or larger is no additional work compared to handling <65k so I implemented it.

PerMalmberg, 2019-07-15:
@squonk11 So the code is now in a state where its possible to send messages of three different sizes from the /websocket.html page, and for each message the server will respond with a hard-coded message. Next step is to add a way to easily receive the data and send data on the same channel from the application. I'm not yet sure how to do that yet.

squonk11, 2019-07-15:
I see you are extremely fast. For me this step would have taken much longer. 
From the application point of view it would be nice to have a class WSHandler with virtual functions like onMessage, onError, onClose, etc. which the user has to override. The parameter passed to onMessage could be a std::streambuf from which the application can read. Maybe you can have a look at [this](https://github.com/fhessel/esp32_https_server) repository in the classes WebSocketInputStreambuf and WebsocketHandler. Feel free to ask if you need details on the solution there. 
But other implementations are also o.k for me if you have better ideas or special requirements.

PerMalmberg, 2019-07-16:
@squonk11 Well, I know my way around the system.

Just tried running it on the ESP32, it crashes when pressing the "Send 65k" button, with the following message, "Send short" button works so not sure what is going on.

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40091cd0  PS      : 0x00060533  A0      : 0x800922c6  A1      : 0x3ffd8a50  
0x40091cd0: is_free at /home/permal/esp/esp-idf/components/heap/multi_heap.c:131
 (inlined by) multi_heap_malloc_impl at /home/permal/esp/esp-idf/components/heap/multi_heap.c:432

A2      : 0x3f8046c8  A3      : 0x000006a8  A4      : 0x3f82c284  A5      : 0x000000fe  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x00000000  A9      : 0xba465551  
A10     : 0x3f82fd80  A11     : 0x0000081c  A12     : 0x3f82fd84  A13     : 0x3f82bc90  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xba465551  LBEG    : 0x400914fc  LEND    : 0x40091512  LCOUNT  : 0xffffffff  
0x400914fc: memcpy at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:146

0x40091512: memcpy at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:161

ELF file SHA256: b128111c039163747f16df4696be3f089a81025c511728f6ae254b877c797ceb

Backtrace: 0x40091ccd:0x3ffd8a50 0x400922c3:0x3ffd8a80 0x40086a99:0x3ffd8ab0 0x40086ba5:0x3ffd8ae0 0x40081d7d:0x3ffd8b30 0x401aa093:0x3ffd8b60 0x401b476b:0x3ffd8ba0 0x4008c9c5:0x3ffd8c20 0x4009847e:0x3ffd8c60
0x40091ccd: multi_heap_assert at /home/permal/esp/esp-idf/components/heap/multi_heap_platform.h:54
 (inlined by) multi_heap_malloc_impl at /home/permal/esp/esp-idf/components/heap/multi_heap.c:431

0x400922c3: multi_heap_malloc at /home/permal/esp/esp-idf/components/heap/multi_heap_poisoning.c:191

0x40086a99: heap_caps_malloc at /home/permal/esp/esp-idf/components/heap/heap_caps.c:111

0x40086ba5: heap_caps_malloc_prefer at /home/permal/esp/esp-idf/components/heap/heap_caps.c:189

0x40081d7d: wifi_malloc at /home/permal/esp/esp-idf/components/esp32/esp_adapter.c:66

0x401aa093: esf_buf_alloc at ??:?

0x401b476b: wdevProcessRxSucDataAll at ??:?

0x4008c9c5: ppTask at ??:?

0x4009847e: vPortTaskWrapper at /home/permal/esp/esp-idf/components/freertos/port.c:143
```

PerMalmberg, 2019-07-16:
Tried to find what caused the error....three hours later....updated ESP-IDF, recompiled and it no longer crashes. I have no idea what the issue was. I ran the https_server/websocket example on my Linux machine with ASAN enabled without any issues.....

squonk11, 2019-07-16:
Since there are many functions concerning memory allocation involved I think there was a problem with the higher amount of data for the >65k. I agree that it is not really satisfying if an error suddenly disappears - it's better to know what happened and what fixed it. But this will be difficult to figure out...

PerMalmberg, 2019-07-16:
Yeah, problems just disappearing are not fun, but I'm going to but it down the the outstanding memory  cache issue logged in ESP-IDF, the same one that likely prevents TLS from functioning on the ESP when compiling with gcc8, which I am.

I'll be pushing an update in about an hour containing a near complete implementation.

PerMalmberg, 2019-07-16:
Think I've found the cause of the crash, fixing that before pushing.

PerMalmberg, 2019-07-16:
@squonk11 Alright, fixed the issue, all my fault. :open_mouth: 

This line shows how to tell the framework what websocket server type to use for the specified API path. You asked for things like on_connect/on_disconnect events, but they aren't needed - when the WebsocketServer (and derived class) is created, that's when the connection has been established, likewise on disconnect, the instance is destructed so the d'ctor is where to do cleanup.

[This line](https://github.com/PerMalmberg/Smooth/blob/feature/49-websocket/test/http_server_test/WSEchoServer.cpp#L32) shows how to send a response to the client.

squonk11, 2019-07-16:
The first "This line shows" in your last post has no link. I guess it is: [this](https://github.com/PerMalmberg/Smooth/blob/feature/49-websocket/test/http_server_test/http_server_test.cpp#L317)? 
As I said I am not so familiar with templates - I need to study your code and play with it a while in order to understand everything.

PerMalmberg, 2019-07-16:
Ops, yes sorry, I missed to add the link.

Here's how it works:

`WSEchoServer` inherits from the `WebsocketServer`, which allows the framework to treat it as a `WebsocketServer`, regardless of what the application programmer puts into it, i.e. regular base classe with polymorphism.

The line you rightly point to tells the framework that any upgrade request made to the `/chat` path will result in the provided `WebsocketServer`, in this case the `WSEchoServer` to be used to handle that specific connection.

PerMalmberg, 2019-07-16:
@squonk11 With the last few adjustment it feels pretty stable. Please give it a try and let me know what you think/find.

squonk11, 2019-07-16:
o.k. I will have a try during the next days.

PerMalmberg, 2019-07-17:
> o.k. I will have a try during the next days.

Cool. I'll be mostly offline for a week, starting Thursday so replies may be slow or none at all during that time.

PerMalmberg, 2019-07-17:
Funny how things can feel stable for so long and then something pops up that totally pulls the rug from under your feet. I started running the websocket stuff over TLS and experienced SIGSEVs. Turns out SecureSocket had an issue. Also, When ASAN was enabled, performance was horrible without enabling optimizations.

PerMalmberg, 2019-07-17:
Last commit for a week. I implemented a working Ping/Pong, but destruction of the WSEchoServer crashes (as expected) since queues are not designed to be destructed once they have been created. There's two solution to this: 1) Use a pool for of the WS servers, or 2) redesign the queues. The first one gives the benefit of being memory static, the seconds one is likely a lot of work but also something I've been wanting to do for a while. I'll have to think on what's the best cause of action.

PerMalmberg, 2019-07-17:
Note to myself. If `QueueNotification` holds std::weak_ptr<ITaskEventQueue>` instead of `ITaskEventQueue* ` in its `queues` member and all `TaskEventQueue` are created on the heap via a factory method (same as the timers are), then `QueueNotification::wait_for_notification()` can check if the queue still is valid via `std::shared_ptr::lock()` which would prevent further use of that queue once it has been destructed.

This looks to be a fairly small change. It breaks all uses of `TaskEventQueue` and `ISRTaskEventQueue`, but I think that it'll be worth it.

squonk11, 2019-07-17:
I am currently starting to test your WSEchoServer. But before actually using WSEchoServer code I am already running into problems a soon as I include the feature/49-websocket branch into the esp-idf components: when I try to compile any existing projects I get the error message 
` Syntax error in cmake code at
      C:/Users/springob/esp/test/build/build_properties.temp.cmake:2
    when parsing string
      C:\Python27\python.exe
    Invalid escape sequence \P`
Obviously there is a problem with setting the path to Python.exe. The cmake files contains the Python path with "\" (windows style) all other paths are using "/" (unix style).
Do you have a hint what is going wrong?

PerMalmberg, 2019-07-17:
No idea, I'm on Linux and have never tried using this on Windows. For reference, here's how I build and run the test projects:

```
cd path/to/root_of_smooth
(select test project in test/CMakeLists.txt)
mkdir build
cd build
cmake .. -G Ninja -DESP_PLATFORM=1 -DCMAKE_TOOLCHAIN_FILE=$IDF_PATH/tools/cmake/toolchain-esp32.cmake -DTARGET=esp32 && ninja clean && ninja -j 8
~/esp/esp-idf/tools/idf.py -C .. -p /dev/ttyUSB1 flash monitor
```

squonk11, 2019-07-17:
Unfortunately I have no performant Linux machine available. I could try with WSL under Windows - but this is just some kind of Linux-emulator and compiling is really slow. I will at least try in order to see if it works this way.

squonk11, 2019-07-17:
at least menuconfig and compilation of existing projects works under WSL. But in WSL I additionally have the problem with accessing the serial interfaces and the esp-idf provided monitor program can not be stopped by Ctrl-] because the key is not recognized. 
I think I have to install a real Linux on an external SSD in order to continue. That will take me some time in order to set everything up correctly (especially my three monitors under Linux might be a problem...).

PerMalmberg, 2019-07-17:
Or install it in a VM under Hyper-V, or Virtual Box?

squonk11, 2019-07-23:
I tried with VirtualBox but the serial interface does not work properly - only one out of five flash attempts work. That is not really satisfying. I also tried to fix the problem under WSL - but also without success. Now I setup your sources inside a "real" esp-idf component using the cmake build system provided by espressif under windows. This now compiles without error.
When flashing it I try to access via http and https. Both have its problems:

1. When I try to access the web_root on the SD-card using HTTP I get the error message 404 and in the monitor I see: `I (137995) ServerSocket: Connection accepted
I (138046) HTTPServer: Request: GET: '/'
I (138049) HTTPServer: Reply: Not Found
W (143063) SocketDispatcher: Receive timeout on socket 0x3fff1c10 (5000 ms)
I (143064) Socket: [, 0, 56, 0x3fff1c10]: Socket stopping
I (143079) Socket: [, 0, -1, 0x3fff1c10]: Disconnected`
Adding "/index.html" to the URL also does not help.

2. When I try to access web_root using HTTPS I get the error message `Secure Connection Failed. An error occurred during a connection to 192.168.1.31:8443. SSL peer reports incorrect Message Authentication Code. Error code: SSL_ERROR_BAD_MAC_ALERT` in the browser and in the monitor I see `I (80137) ServerSocket: Connection accepted
E (82301) SecureSocket: mbedtls_ssl_handshake_step returned -29056: SSL - Verification of the message MAC failed
I (82302) Socket: [, 0, 56, 0x3fff1f64]: Socket stopping
I (82308) Socket: [, 0, -1, 0x3fff1f64]: Disconnected`

Do you  have any hints?

PerMalmberg, 2019-07-23:
1.  Did you copy anything to the Web root folder on the SD card?

2.  Yes, that's an outstanding issue in IDF, should be linked in another
issue in this repo.

squonk11, 2019-07-23:
ad 1: yes, I copied the contents of [this](https://github.com/PerMalmberg/Smooth/tree/master/test/http_server_test/web_root) into the folder "web_root" on the SD card. I also tried to copy this contents into the root folder and into "sdcard/web_root" - but nothing helps.

squonk11, 2019-07-23:
I see that you activate the following handlers in the end of main:
`        insecure_server->on(HTTPMethod::GET, "/api/blob", blob);
        insecure_server->on(HTTPMethod::POST, "/upload", upload);
        insecure_server->enable_websocket_on<WSEchoServer>("/echo");
`
maybe I need to add a handler to "/" ?

PerMalmberg, 2019-07-24:
Is the SD card getting mounted properly? It should work as-is, no need to
add handler for "/".

squonk11, 2019-07-24:
yes, the SD card gets detected properly. During the boot there is a message saying that it works with 20MHz and it gives the correct size.
Maybe this evening I could try to read and write the SD card using the functions provided by your framework. If this works the problem is somewhere else.

PerMalmberg, 2019-07-24:
How is the SD card formated? FAT32?

Also make sure you have long filenames enabled in menuconfig.

squonk11, 2019-07-24:
yes, long filenames could be the issue. I remember that 8.3 filenames are configured by default and I did not change this setting in menuconfig. I will check this this evening but probably this is the issue.

PerMalmberg, 2019-07-24:
If you're using the sdkconfig that comes with smooth it should have it
enabled....

squonk11, 2019-07-24:
unfortunately I was not using the sdkconfig of smooth because I was not sure if it is 100% compatible with esp-idf. So I setup a new one. During this process I also had to set some configurations concerning stack of main() and memory for the TLS data (using PSRAM) because I encountered some errors due to that.

squonk11, 2019-07-24:
The long filenames were the problem. But I then had problems loading the complete page - the ESP32 crashed while loading the icons. Then I compared the sdkconfig of Smooth to mine and made some corrections. Now the page loads without problems.
When testing the Websockets I encounter the problem that e.g. for 65k not the total amount of data is transferred and the websocket connection gets disconnected. I will check the sdkconfig again.

PerMalmberg, 2019-07-24:
So what happens when not all data is loaded? I've not seen any such
problems...doesn't mean there are none though...

squonk11, 2019-07-24:
I must correct myself: I checked with wireshark and I see that the data fully arrives. After the transmission the websocket connection is being closed - this is what I did not expect. I see that the connection is not being closed by the client (browser); so, the framework closes it. I would expect that the connection persists until a timeout elapses or until I click the "disconnect" button.

PerMalmberg, 2019-07-24:
I'd expect that too, and that is what I have experienced. I've been
clicking those buttons until they wore out, trying to break things...
What's in the log output?

squonk11, 2019-07-24:
Now I checked the console log. There is the error message: `The connection to ws://192.168.1.31:8080/echo was interrupted while the page was loading.` This happens after 4096 bytes ("a") were received. Then the connection is disconnected. Is this behavior correct?

PerMalmberg, 2019-07-24:
That's the client log, what does the esp log?

squonk11, 2019-07-24:
The ESP log is:
`I (2785006) ServerSocket: Connection accepted`
`I (2787370) Socket: [, 0, 56, 0x3ffed474]: Socket closed`
`I (2787371) Socket: [, 0, 56, 0x3ffed474]: Socket stopping`
`I (2787375) Socket: [, 0, -1, 0x3ffed474]: Disconnected`
no error message.

PerMalmberg, 2019-07-25:
Hm, will see if I can reproduce it when I get home this evening or
tomorrow.

squonk11, 2019-07-25:
o.k. for me there is not hurry - currently it is a private project.
One additional information: I am using Firefox. Shall I provide a wireshark file?

PerMalmberg, 2019-07-25:
Attach it, might be useful.

PerMalmberg, 2019-07-25:
Not sure I can reproduce the issue you specify, but I do see the following:

```
W (49802) wifi: alloc eb len=24 type=3 fail, heap:4021356

W (49803) wifi: m f null

W (49813) wifi: alloc eb len=24 type=3 fail, heap:4019276

W (49814) wifi: m f null
```

This means the the wifi driver can't allocate memory as per #3592. When this happens, I do get disconnects (and the device is no longer reachable).

PerMalmberg, 2019-07-26:
@squonk11 I think I get the same error you get now, but only when running on the ESP:

> The connection to ws://192.168.10.94:8080/echo was interrupted while the page was loading.

In #53 I added a reason message for all calls to `ISocket::close()` and it now prints `Underlying socket closed (recv returned 0)` which leads me to believe it is initiated on the client side. This has to be verified, but nothing in Smooth is actively closing the socket. I guess Wireshark can tell us who sends the RST signal?

PerMalmberg, 2019-07-26:
@squonk11 I ofc meant who sends the first TCP FIN flag. It seems to me it is sent from the browser (my ESP IP is 192.168.10.94)

[image]

What I I don't understand is why it seemingly only happens when sending large >65k messages?

I've attached the Wireshark capture. Thoughts?

[socket_close.zip](https://github.com/PerMalmberg/Smooth/files/3435853/socket_close.zip)

PerMalmberg, 2019-07-26:
This is a capture of a 125 byte message. It doesn't close the socket.
[socket_close_125.zip](https://github.com/PerMalmberg/Smooth/files/3436236/socket_close_125.zip)

squonk11, 2019-07-26:
When I check your wireshark capture "socket_close" I see that you first sent three times 125 byte packages which are sent back correctly. After that you probably switched to the 64k packet and try to send that. But here I see that these are not sent as "Websocket" protocol but as "TCP". It looks as if the client makes the mistake... ?

squonk11, 2019-07-26:
I think I need to correct myself: probably the websocket transmission of lager size over several packets is done with TCP packets and only the last packet is marked as "Websocket". Unfortunately I am not sure about that because in my previous project I was only using packets smaller that 1400 bytes.

squonk11, 2019-07-26:
Checking the Wireshark capture I see the following peculiarities:

1. The client (browser) is sending as websocket frame: 0x81,0xfe,0xff,0xff. That means it is setting the FIN bit and the MASK bit and opcode 0x01. Then it sends 64k data.
2. The server is sending the data in chunks of 4096 byte, probably due to the fact that the ESP has not enough memory to store the complete 64k data.
3. The first data chunk is started with a websocket frame of 0x81,0x7e,0x10,0x00. That means it is setting the FIN bit and opcode 0x01. Then it sends data: all "a".
4. The second data chunk is started with the same websocket frame as in 3. and send data: all "a"
5.  The third data chunk is started with a websocket frame of 0x01,0x7e,0x10,0x00. That means this time the FIN bit is not set and the opcode is 0x01. The data again is all "a"
6. The fourth data chunk is started with 0x00,0x7e,0x10,0x00. This time FIN not set and the opcode is 0x00 (continue) this time. This time the data is **not** all "a" but some other data. Maybe the pointer to the data is corrupted on the ESP side?
7. The sixth up to the 16th data chunk are sent with a websocket frame of  0x00,0x7e,0x10,0x00 and the data still is not all "a"
8. The 17th chunk starts with 0x80,0x7e,0x10,0x00, so FIN bit set and opcode 0x00 (continue).

I think the handling of the FIN bit and the opcodes is not properly selected. In addition to that the data which is being sent by the server is not the data which it should be.

PerMalmberg, 2019-07-26:
Hm, I'll have a look at the handling of FIN and control codes.

> The server is sending the data in chunks of 4096 byte, probably due to the fact that the ESP has not enough memory to store the complete 64k data.

The example server is configured to handle all data in 4096 blocks to keep memory usage constant and deterministic. This can be set on a per-server basis.

[41 more replies]

closed 2019-08-07 by PerMalmberg