PerMalmberg/Smooth #64: CMake Failed to resolve component 'smooth_component'.

COM8, 2019-08-19:
Probably related to: Build system #60

When I take the `hello world` example program from ESP-IDF and update the `CMakeLists.txt` files I get the following when trying to compile via `idf.py build`:
```
$ idf.py menuconfig
Checking Python dependencies...
Python requirements from /<ESP32-PATH>/esp-idf/requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory /<ESP32-PATH>/esp32-xmpp-iot/hello_world/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 /<ESP32-PATH>/esp32-xmpp-iot/hello_world"...
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.21.0") 
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /<ESP32-PATH>/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /<ESP32-PATH>/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /<ESP32-PATH>/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /<ESP32-PATH>/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /<ESP32-PATH>/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: 85b49e5-dirty
-- Building ESP-IDF components for target esp32
CMake Error at /<ESP32-PATH>/esp-idf/tools/cmake/build.cmake:185 (message):
  Failed to resolve component 'smooth_component'.
Call Stack (most recent call first):
  /<ESP32-PATH>/esp-idf/tools/cmake/build.cmake:211 (__build_resolve_and_add_req)
  /<ESP32-PATH>/esp-idf/tools/cmake/build.cmake:407 (__build_expand_requirements)
  /<ESP32-PATH>/esp-idf/tools/cmake/project.cmake:341 (idf_build_process)
  CMakeLists.txt:14 (project)

-- Configuring incomplete, errors occurred!
See also "/<ESP32-PATH>/esp32-xmpp-iot/hello_world/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

My `./CMakeLists.txt`:
```CMAKE
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.10)

set(CMAKE_CXX_STANDARD 17)

if(${ESP_PLATFORM})
include($ENV{IDF_PATH}/tools/cmake/project.cmake)

# Include Smooth as a component
set(EXTRA_COMPONENT_DIRS
         externals/smooth/smooth_component)

project(hello-world)
else()
    # Empty project when not building for ESP (i.e. when loading the project into an IDE with already configured tool chains for native Linux)
endif()
```

My `./main/CMakeLists.txt`:
```CMAKE
cmake_minimum_required(VERSION 3.10)

set(CMAKE_CXX_STANDARD 17)

set(SOURCES hello_world_main.cpp)

idf_component_register(SRCS ${SOURCES}
        INCLUDE_DIRS
            ${CMAKE_CURRENT_LIST_DIR}
            $ENV{IDF_PATH}/components
        REQUIRES
            smooth_component
        )
```

PerMalmberg, 2019-08-19:
Hello Fabian,

I'd say that this line doesn't match up to where your copy of Smooth is placed.

```cmake
set(EXTRA_COMPONENT_DIRS
         externals/smooth/smooth_component)
```

So, is your smooth-root at `<project-path>/externals/smooth` ?

COM8, 2019-08-19:
That's it thanks!  
I added `smooth` to the `esp-idf` repo as submodule not my actual project.

closed 2019-08-19 by COM8