PerMalmberg/Smooth #66: Syntax error in cmake code at ... when parsing string

KerryRJ, 2019-08-29:
OS:- Windows 10 1903
ESP:- IDF 4.1
IDE:- Visual Studio 2019 Community 16.3.0 Preview 2

I get the following error when compiling a project using Smooth under externals\smooth\smooth_component.

[CMake]   Invalid character escape '\D'.
[CMake] Call Stack (most recent call first):
[CMake]   C:/Dev/Repo/Github/esp-idf/tools/cmake/component.cmake:450 (__component_add_include_dirs)
[CMake]   externals/Smooth/smooth_component/CMakeLists.txt:23 (idf_component_register)
[CMake] -- Configuring incomplete, errors occurred!

The problem is the path separator in Windows.

Changing line 26 from $ENV{IDF_PATH}/components to ${IDF_PATH}/components allows the generation of the CMAKE cache to complete successfully. Is there a better solution?

PerMalmberg, 2019-08-29:
It's meant to add include directories to IDF components to whereever IDF is is placed. To do that it looks for the environment variable IDF_PATH so `$ENV{IDF_PATH}` is the correct syntax. I think that when you change it `${IDF_PATH} you get include directories to just `/components` (unless IDF_PATH is set elsewhere ofc).

What does the following print if you add it on line 22, i.e. what is your environment variable set to?

`message(FATAL_ERROR ${IDF_PATH}`

KerryRJ, 2019-08-30:
It prints the normalized IDF SDK path - C:/Dev/Repo/Github/esp-idf.
If I use the message(FATAL_ERROR $ENV{IDF_PATH}) then it prints the Windows path separator - C:\Dev\Repo\Github\esp-idf and this causes CMAKE to fail.

Lines 8 - 9 of idf.cmake sets the variable ${idf_path} and normalizes it.
`    set(idf_path $ENV{IDF_PATH})
    file(TO_CMAKE_PATH "${idf_path}" idf_path)
`

PerMalmberg, 2019-08-30:
Ok.

Quick fix: Change your enviroment variable to use forward slashes.
Real fix: Adding normalization to the build script.

PerMalmberg, 2019-08-30:
See #72

closed 2019-08-30 by PerMalmberg