PerMalmberg/Smooth #76: Smooth does not like ESP-IDF Bluetooth

COM8, 2019-09-03:
When you like to use `Bluetooth` in combination with `Smooth` it will result in:
```
fatal error: esp_bt.h: No such file or directory
#include "esp_bt.h"
              ^~~~~~~~
```

How to reproduce:
1. Use the [gatt_server_service_table]() example
2. Add `Smooth` as component
3. `idf.py menuconfig` (Enable C++ extensions, set tick rate to 1000)
4. `idf build` or `cmake .. -DESP_PLATFORM=1 -DCMAKE_TOOLCHAIN_FILE=$IDF_PATH/tools/cmake/toolchain-esp32.cmake && ninja` and `make`
4. :boom:

PerMalmberg, 2019-09-03:
In which file?

COM8, 2019-09-03:
```
gatt_server_service_table/main/gatts_table_creat_demo.c:26:11:
fatal error: esp_bt.h: No such file or directory
  #include "esp_bt.h"
           ^~~~~~~~~~
```

PerMalmberg, 2019-09-03:
Can you provide a test project that exhibits this problem?

COM8, 2019-09-04:
Sure: https://github.com/COM8/Control-Test/tree/esp-idf-bluetooth-smooth

```
git clone git@github.com:COM8/Control-Test.git
git checkout esp-idf-bluetooth-smooth
git submodule update --init --recursive
cd gatt_server_service_table
idf build
```

COM8, 2019-09-04:
I did a step by step setup if you have a look at all commits.

PerMalmberg, 2019-09-04:
Well, there is no file named `esp_bt.h` in the main directory so I'm not surprised it can't find it.

That said, i think it comes down to how the IDF CMake build system works. 

`idf_component_register(SRCS ${SOURCES}
                       INCLUDE_DIRS ${CMAKE_CURRENT_LIST_DIR} $ENV{IDF_PATH}/components
                       REQUIRES smooth_component)`

The above only specifies smooth_component as a required component and thus IDF won't add include paths to any other components. So if you add <name of BT> I think it'll work. 

I'm at work now so I can't test this theory.

COM8, 2019-09-06:
OK fixed it by adding `bt`. The result looks like:
```CMAKE
idf_component_register(SRCS ${SOURCES}
                       INCLUDE_DIRS ${CMAKE_CURRENT_LIST_DIR} $ENV{IDF_PATH}/components
                       REQUIRES smooth_component
                       bt)
```
Thanks for your help!

closed 2019-09-06 by COM8