PerMalmberg/Smooth #81: License Model

jeremyjh, 2019-09-05:
Hi, this looks like a really nice library and the style suits my tastes well. Unfortunately, it is impossible to use it in a commercial product under the GPL license. If I base my hobby project on this and later decide it has commercial prospects I'd have to first rewrite it, unless there is a dual-license model. Do you intend to sell a commercial license as well?

PerMalmberg, 2019-09-05:
Hi @jeremyjh 

Glad you like it, and thank you for actually caring about the license.

I've written Smooth mainly because I need it, not to make money of it. As such, I've had no thought on a dual-license. I did however explicitly change from MIT to GPL for the reason that I didn't want anyone to just runoff with it without ensuring that they would be obligated to contribute to back to the project - at least on paper.

I have since then given some thought to changing it back to MIT. Since I have no way of knowing if someone violates the GPL, what benfits do the GPL really give me? It just prevent honest people like yourself from using my work.

Perhaps most people don't think twice about offering monetary compensation if they can just take the code because the license allows it, such as with MIT. On the other hand, if they are honest and the GPL prevent usage, then there's no chance what so ever of compensation, right?

As you likely understand by know, I've not made up my mind yet as to what license model I want to use. Please help me decide, fellow developer.

INAL, but if I'm not mistaken, the two current contributors (ping @squonk11 & @KerryRJ) with actual code in the repository would have to sign off on a license change since they contributed the code under the GPL. If they won't, then their commits would have to be removed and rewritten.

jeremyjh, 2019-09-05:
I understand the reasoning, but I think GPL is more suited to end-user software or standalone server programs than to libraries. If Emacs is GPL, I can still use it at work to edit my own, or my company's source code without fear that what I create with it has become a derivative work with copyright obligations to the Emacs developers. The same is not true for libraries, which create a derivative work of any software linked to them. This is why the LGPL was created. MIT and Apache 2.0 are good as well; personally I prefer the latter due to the patents clause. Anyway, that's my two cents.

KerryRJ, 2019-09-06:
Let me know when and how to sign off once the decision has been made.

squonk11, 2019-09-06:
Also for me it is no problem to put my small contribution under a different / more permissive license model. Since also I am interested to maybe use it in a commercial project a more permissive license would be helpful. 
But as already mentioned this commercial product is just a service tool which is not a part of the companies intellectual property - and so I would not have a problem publishing the whole source code of this service tool.

PerMalmberg, 2019-09-06:
Glad to hear that, @squonk11 and @KerryRJ 

The collective wisdom of the internet says that contributors can simply agree to the license change in a PR, though not exactly how. An actual commit, or just a comment here on GitHub?

PerMalmberg, 2019-09-07:
Alright - there is now a branch named new-license in which I've updated everything to Apache 2.0.

@squonk11 and @KerryRJ could you please fork from that branch, update CONTRIBUTORS.md and add yourselves to the contribution table, using a commit-comment like this: 

    "By this commit I authorize re-licensing of my contributions to the Apache 2.0 license."

INAL, but I think that should be sufficient.

Here's a suggestion on what to enter into the table:

`|[KerryRJ](https://github.com/KerryRJ)|[PR#71](https://github.com/PerMalmberg/Smooth/pull/71)|`

`|[squonk11]([https://github.com/squonk11)|[PR#65](https://github.com/PerMalmberg/Smooth/pull/65)|`

Once that is done I'll merge everything back into master.

PerMalmberg, 2019-09-09:
@squonk11 has done his PR. Now we're just waiting for @KerryRJ and the license change can be completed.

PerMalmberg, 2019-09-11:
@KerryRJ Thanks for the PR.

I've now merged the new-license branch into master.

PerMalmberg, 2019-09-11:
For future references, these are the commits where contributors authorizes the license change:

https://github.com/PerMalmberg/Smooth/commit/f76c3afbc99c8224bd3aedbb5b06bfcb335aa79c
https://github.com/PerMalmberg/Smooth/commit/a6150bf75507ef4584625212e2412bac1f4857bc

jeremyjh, 2019-09-11:
Awesome, thank you all so much!

closed 2019-09-11 by jeremyjh