PerMalmberg/Smooth #99: SPI still available?

enelson1001, 2019-10-02:
I noticed that lib/smooth/include/smooth/io does not contain spi.  Have you removed support for SPI?

PerMalmberg, 2019-10-02:
What was there was never a complete implementation, and even less working.

enelson1001, 2019-10-02:
OK, I'll see if I create something on my own.  I will probably combine
Master and Device into one class since I have only one spi device.

PerMalmberg, 2019-10-03:
I don't remember the API exactly, but weren't there two sides of it? Merging them into one sounds like creating a God-object...

squonk11, 2019-10-03:
Just a hint from my side: the slave mode of the ESP32 has a design flaw. Due to this there is the following restriction: ` master should write lengths which are a multiple of 4 bytes. Data longer than that will be discarded`; see here: [known issues](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_slave.html#restrictions-and-known-issues).

enelson1001, 2019-10-03:
@Lothar Thanks for the hint.
@Per After rethinking will go back to yuor designa approach and see what I
come up with.  Will probably have more questions later.

PerMalmberg, 2019-10-03:
Remembered that I have have e-paper displays that work over SPI. Not sure how much time I have left over in the near future but I'm up for some discussions around this.

enelson1001, 2019-10-04:
Roadblock #1
1. git clone  https://github.com/PerMalmberg/Smooth.git smooth
2. git submodule update --init --checkout --recursive
3. git checkout -b eds_smooth
4. modified files.cmake to include core/io/spi header files. These files
came from repository I downloaded last year in November.
5. added the header files under include/smooth/core/io/spi.  ISPIdevice.h,
Master.h, SPIDevice.h
6. did git add lib/files lib/smooth/include/smooth/core/io/spi/
7. did git -m "added spi headers"
8. now when I run the build (using VSCcode) the build stops at the very
beginning and hangs.  Really screws up VSCode as something hogging the CPU.
9. if I remove the lib/smooth/include/smooth/core/io/spi/ the build stops
because it can't find Master.h (which is expected) but doesn't hog the cpu.
10. does this occur on your development system?

Roadblock #2
1. Is there a way to structure the spi device so we can use the
transaction_t "t.user" for a variable that tells us the state the DC pin
should be instead of holding "this".
2. I am trying to do a driver for the ili9341.  If you look at
https://github.com/espressif/esp-idf/blob/master/examples/peripherals/spi_master/main/spi_master_example_main.c
Line 290 - send lines().  The author was able to queue a bunch of
transactions but each transaction would use "user" to designate if the DC
pin was to be high or low.

PerMalmberg, 2019-10-04:
1. I have no idea, but I don't remember it doing that when I wrote the code.

2. Are you saying my old code you dug up uses .user to to point to the driver itself? 

I suppose what you're really after is the ability to set the DC line in this function?

```c++
void lcd_spi_pre_transfer_callback(spi_transaction_t *t)
{
    int dc=(int)t->user;
    gpio_set_level(PIN_NUM_DC, dc);
}
```

I've litterally just given it 3 minutes of thought, but if you can use `t` as a key, you could do a lookup for the DC value via the driver instance. Just have to keep in mind that the function is called in ICQ-context.

enelson1001, 2019-10-05:
I was able to get past roadblock #1 and am able to compile ST7735 if I
commented out "set_compile_options(${COMPONENT_LIB})" in
smooth-component/CMakeLists.txt otherwise I was getting following errors.

/home/ed/esp32-idf/esp-idf/components/esp_rom/include/esp32/rom/lldesc.h:153:18:

error: conversion from 'uint32_t' {aka 'unsigned int'} to 'volatile
unsigned
char:1' may change value [-Werror=conversion]
[build] ds->owner = owner;

As far as roadblock #2 I don't see how to accomplish with the current
implementation of SPIDevice.  I really need transaction.user to include
both "this" and the state I want the DC pin to be.  This  way I can queue
transactions and each transaction has the knowledge on what the sate the dc
pin should be.  I need "this" to be able to call
pre_transmission_action(trans) but need the pre_transmission_action to
include the state I want the DC pin so then pre_transmission_action can set
the dc pin.  I will keep think about it but am stumped for the moment.

PerMalmberg, 2019-10-05:
1. You should leave that line active it enables sane warning options. A better options is to do like on line [24 and 28](https://github.com/PerMalmberg/Smooth/blob/9c59426eee61fde76aa4717b3426b1c7dc04a911/lib/smooth/include/smooth/core/filesystem/SDCard.h#L24) for that warning.

2. You have the entire array of transactions currently being sent available in the object pointed to by `this`, right? If so, wrap each transaction, something like this:

```c++
struct MyTransaction
{
    spi_transaction_t t;
    bool dc_state;
}
```

In the driver:
```c++
MyTransaction data_to_send[N];
```

For each transaction, make the `user`pointer point to itself:
```c++
data_to_send[i].t.user = &data_to_send[i];
```

Then, we can do this:
```c++
void lcd_spi_pre_transfer_callback(spi_transaction_t *t)
{
    auto transaction = reinterpret_cast<MyTransaction*>(t->user);
    gpio_set_level(PIN_NUM_DC, transaction->dc_state);
}
```

Big disclaimer, this is just brain dump :)

PerMalmberg, 2019-10-05:
Actually, I presume you need an array of `spi_transaction_t` so you can't wrap it. So instead place dc_state as a separate array:

In the driver:

```c++
std::array<bool, N> dc_state{};
std::size_t current_transaction{};
```

For each transaction, make the `user`pointer point to the driver
```c++
data_to_send[i].user = this;
```

Then, we can do this:
```c++
void lcd_spi_pre_transfer_callback(spi_transaction_t *t)
{
    auto driver = reinterpret_cast<Driver*>(t->user);
    gpio_set_level(PIN_NUM_DC, driver->dc_state[driver->current_transaction]);
    driver->current_transaction++;
}
```

Big disclaimer, this is just brain dump :)

enelson1001, 2019-10-05:
Wish I new better how to format this message with code.

Pseudo code from
https://github.com/espressif/esp-idf/blob/master/examples/peripherals/spi_master/main/spi_master_example_main.c
Lines 290 to 338

void send_lines()
{
   spi_transaction_t trans[6];
   trans[odd] has trans.user set to 0 so dc=0
   trans[even] has trans.user set to 1 so dc=1

   now queue all 6 transactions
   spi_device_queue_trans(spi, &trans[], portMax_delay)
}

So I am using the queue_trans to send queued transactions in single write.

This is what I am trying to accomplish.

enelson1001, 2019-10-05:
And using pre_trans_callback to set DC

PerMalmberg, 2019-10-05:
(use regular markdown syntax to format code etc)

Yes, I think my second suggestion should work for you.

enelson1001, 2019-10-05:
OK, I will work with your second suggestions using array of bool and see
what I come up with.  Thanks for your help.

enelson1001, 2019-10-07:
Getting very close now but have some questions.
1. I have implemented the following functions in SPIDevice.cpp is this
where you planned on implementing them?
    bool polling_write(spi_transaction_t& transaction)
    bool queue_transaction(spi_transaction_t& transaction)
    bool wait_for_transaction_to_finish(spi_transaction_t& transaction)
2. Both bool queue_transaction(spi_transaction_t& transaction) and bool
wait_for_transaction_to_finish(spi_transaction_t& transaction) require an
additional parameter called delay.
    a. Should these two functions be modified to include this additional
parameter?
    b. What type should the parameter be?
    c. For the ILI9341 they set delay to portMAX_DELAY what is the
equivalent in smooth?
3. For ILI9341 they suggest setting max_transfer_sz on SPI Master, so
should Master.cpp be modified to include this bus_config.max_transfer_sz?

PerMalmberg, 2019-10-07:
Glad to hear you're making progress. :+1: 

1. I honestly don't remember; I never got very far with the original SPI implementation since I went with all I2C in on the hardware side for my project. Does it make sense to place the methods there? Or, in other words, does it feel natural to do `my_device.queue_transaction?` Is the code up on GH so I can look at it?

2. What's it used for? iirc, portMAX_DELAY means infinite timeout, i.e. blocking operation. I've tried to use `std::chrono::milliseconds` for timeouts in Smooth when possible.

3. If its needed, is there another way? what does it do?

enelson1001, 2019-10-07:
1. I think it make senses since you placed those function calls in
ISPIDevice.h as virtual. And it makes sense to do
my_device.queue_transaction, my_device.wait_for_transaction_to_finish.

2. From Espressif API
     a. *esp_err_t spi_device_queue_trans(spi_device_handle_t handle,
spi_transaction_t *trans_desc, TickType_t ticks_to_wait)* where
ticks_to_wait = Ticks to wait until there’s room in the queue; use
portMAX_DELAY to never time out.

     b. *esp_err_t spi_device_get_trans_result(spi_device_handle_t handle,
spi_transaction_t **trans_desc, TickType_t ticks_to_wait)* where
ticks_to_wait = Ticks to wait until there’s a returned item; use
portMAX_DELAY to never time out.

     c. So if we do add the delay parameter should we leave the type as
TickType?

3. For SPI Master from Espressif API
     a. max_transfer_sz = Maximum transfer size, in bytes. Defaults to 4094
if 0.
     b. For the ILI9341 we would set max_transfer_sz to
Horizontal_Resolution *40(lines)*2(buffers).  For portrait display that
would be 320*40*2 = 25600, for landscape display that would be 240*40*2 =
19200. So yes I think it is required especially for SPI interface to
displays.

I think I will modify and place the modified smooth spi code on git hub
today or tomorrow for your review.

PerMalmberg, 2019-10-07:
Sounds like things are falling into place.

Look in I2CMasterDevice.h for the `to_tick` function for converting from ms to ticks.

enelson1001, 2019-10-07:
Great that looks like it will work.

Now for the last piece.
Long version
I think I have the ILI9341 driver code 99% done (just need to test).  I am
currently developing on the M5Stack.  I have used smooth + LittlevGL +
LittlevGL/drv (spi driver for ili9314) with success but now want to
implement the ILI9341 using the smooth/spi library.  So I making a class
called LvglDispDrv() that runs in a task and uses LittlevGL +
smooth/ILI9341. Inside LvglDispDrv is a method called displ_flush that
calls methods in smooth/ILI9341.  The problem is LittlevGL wants a callback
to the displ_flush method and of-course the compiler doesn't like the
callback to a class method.  I could make the displ_flush method static but
since I am using smooth/ILI9341 methods inside displ_flush I would need to
make those methods static which I really don't want to do.
Short version
Do you have an example how to wrap a static callback method that will call
a class function callback?

enelson1001, 2019-10-08:
Never mind was able to figure it out.  Will post code on gitub in a few
days.

enelson1001, 2019-10-08:
I would like to create a GuiThread (task) and then have additional classes
I create (that are only used for by the gui) run on the GuiThread.  I
looked at Task.h and Task.cpp and Task.h describes a constructor to use to
attach to an exist thread but it show "main" as the thread name to attach
too and I was confused on how to use.  Is there an example of how to
accomplish what I am look for?

PerMalmberg, 2019-10-08:
`core::Task` has two uses: As a standalone task, or to run as the main task (which it was it does when used via the `App` class.

If you just want to create a new Task, then look at `hello_world` (or nearly any other example) which creates a secondary worker task.

enelson1001, 2019-10-10:
I have gotten the program to work with the M5Stack.  Give me a few more
days to clean up the code and create a demo program.  Once I have put on
github I will let you know.

PerMalmberg, 2019-10-10:
Can SPI list IDs of devices on the bus, like I2C can? If so, that'd be a nice and simple demo/test project.

PerMalmberg, 2019-10-11:
FYI: There are two major changes in master now:
- Removal of names for queues.
- New log framework, {fmt}, https://fmt.dev

enelson1001, 2019-10-17:
I think I have finally managed to read registers on the ILI9341.  The ID
register reports 0x00 for the data instead of any useful information, maybe
because they are displays are from China. But I have found other registers
I can read that provide correct results and shows the read is working. I
have not looked at the new changes in master yet.  I plan on trying to hook
up the BME280 sensor (since it also has a spi interface) on the same bus as
the ILI9341 to see if I can communicate with both.  I have family visiting
for next few days so you won't be hearing from me for at least 7 days. So
please don't close out issue yet.

enelson1001, 2019-10-28:
I am ready for you to review the changes I have made and have written a
test program to test two devices on a single SPI bus.  Could you visit
https://github.com/enelson1001 and look at the following repositories when
you have time.
   1. Smooth-SPI-Changes Repository - Files and folders laid out for smooth
with brief description of changes and or additions.
   2. Smooth-Test_Multi_SPI_devices Repository - Test program for testing
two spi devices on single SPI bus.
   3. M5Stack-Smooth-LvGL Repository - Demo using Smooth + LittlevGL
running on M5Stack

Let me know how you want to proceed.  I will be happy to make any changes
you request.

PerMalmberg, 2019-10-28:
Can you please fork the original repo and create a PR for things going into Smooth? That'll clearly show what changes you've made.

Is it possible to use the same relation between spi::Master and an spi-device like there is between i2c::Master and i2c::MasterDevice, where the Master acts like a factory for the device?

Re. BME280. As it seems to be a near copy of the ic2BME280, I think they should either share a base class, or use a shared helper class that performs the calculations etc. Also, there were recently some bug fixes to the i2c::BME280 implementation, make sure those are included of they aren't already.

enelson1001, 2019-10-30:
1. OK will fork original repo when I finish your suggested changes.
2. I have mimicked your i2c approach for spi implementation.
3. For the BME280 I was thinking of making a base bme280 class like you
suggest that would have everything included from your original BME280 class
plus 2 virtual methods, one for virtual method for writing and one virtual
method for reading.  I would then create a spi-bme280 class that inherits
the bme280 and implements it version of reading and writing.  I would
create a i2c-bme280 class that inherits the bme280 and implements it
version of reading and writing.  For spi-bme280 I would also need to
inherit SPIDevice and for i2c-bme280 I would need to inherit
i2cMasterDevice.  Are you OK with multiple inheritance or can you suggest
another way.  For the base bme280 class I was thinking of placing it in a
folder called sensor inside application or do you have some other
suggestion on were the base bm280 class should be placed.

PerMalmberg, 2019-10-30:
You could also place all the common things in a `BME280Common` or some such, that'd avoid the multiple inheritance. Where to place it...well, I think it's fine if it lives in either i2c or spi, no need to create a directory just for that.

closed 2020-01-02 by PerMalmberg