PerMalmberg/Smooth #101: Non-reentrant Task?

jeremyjh, 2019-10-05:
Sorry to raise an issue just for a question, but if I have a Task that has (for example) a 4ms tick interval but sometimes takes 5ms to run, then I believe tick() could get pre-empted and re-entered even if I've set a core affinity, is that right? If the code in tick() is not thread-safe should I just use my own mutex with a try_lock in the task to prevent being re-entered and the queue building up?

PerMalmberg, 2019-10-05:
> Sorry to raise an issue just for a question

No worries, happy to help.

> task to prevent being re-entered and the queue building up

If you have a look at the implementation, you can see that each Task is served by a single thread and [Task::exec()](https://github.com/PerMalmberg/Smooth/blob/master/lib/smooth/core/Task.cpp#L165) directly calls your [tick()](https://github.com/PerMalmberg/Smooth/blob/master/lib/smooth/core/Task.cpp#L165) override, there is no chance of a build up tick "events". There is no re-entrancy issues in `tick()` (unless you start calling it yourself).

[This check](https://github.com/PerMalmberg/Smooth/blob/master/lib/smooth/core/Task.cpp#L165) ensures that the tick has priority over incoming events from queues so that that the tick stays alive and as runs as close as possible to the set interval if the ticks takes longer than expected, or if there is a continuous stream of incoming events on a queue. This prioritization was a conscious choice I made and has worked well for me so far. The rationale is that it is more important that each task can perform its operations on time, than to serve incoming events.

HTH.

jeremyjh, 2019-10-05:
Perfect, thanks!

closed 2019-10-05 by jeremyjh