PerMalmberg/Smooth #103: gpio_install_isr_service(0) in EarlyInit collides with Uart Driver

squonk11, 2019-10-10:
I am currently implementing a serial communication via Uart with smooth library working in parallel. Unfortunately I was running into problems with sent serial data being corrupted. After some investigation I found out that I can make it work if I comment out `gpio_install_isr_service(0);` in EarlyInit ctor. Obviously this seems to collide with the uart driver of ESP-IDF. Is this initialization `gpio_install_isr_service(0);` really needed in this place? For which features of smooth it is needed?

PerMalmberg, 2019-10-10:
Its there to initialize the ISR service so that, for `InterruptInput` and `InterruptInputCB` can use `gpio_isr_handler_add()`. Perhaps a better design is to have then - if not previously done so - install the ISR service.

squonk11, 2019-10-10:
Is it possible to remove 'gpio_install_isr_service(0);' from EarlyInit and do it when it is needed for IOs?

PerMalmberg, 2019-10-10:
It's possible. Like I said, a better design is likely to have _InterruptInput_ and _InterruptInputCB_ call it, if it hasn't already been done (taking care of threading issues).

One of the reasons for it to be in EarlyInit is that its called before any other task, assuming you kick things off via smooth::App.

PerMalmberg, 2019-10-11:
FYI: There are two major changes in master now:
- Removal of names for queues.
- New log framework, {fmt}, https://fmt.dev

squonk11, 2019-10-13:
if I do `gpio_uninstall_isr_service()` in the beginning of my code the uard driver works. I can use this as a workaround for the moment.

squonk11, 2019-11-02:
I think this topic can be closed.

PerMalmberg, 2019-11-02:
I'll do it, but I think you can close issues yourself?

closed 2019-11-02 by PerMalmberg