PerMalmberg/Smooth #111: Question: how to setup simple TCP/IP communication?

squonk11, 2019-10-24:
I am currently trying to setup a simple TCP/IP communication. In order to do that I am using the "server_socket_test" as a basis. 
After compiling and flashing the sample program"server_socket_test" I see that ESP32 gets an IP address. When I send a TCP/IP telegram to the ESP (e.g. using [PacketSender)](https://packetsender.com/) I see that a TCP/IP connection is being build up and directly closed again (after approx. 500ms). But I don't see any logging of the data sent and I also don't see that a "DataAvailableEvent" is being triggered in the StreamingClient.
This is the log I see:
```I (4005) SocketDispatcher: Network up, sockets will be restarted.
I (4005) tcpip_adapter: sta ip: 192.168.1.31, mask: 255.255.255.0, gw: 192.168.1.1
I (4007) Socket: [0.0.0.0, 8080, 54, 0x3ffbd808]: Created server socket
I (32465) ServerSocket: Connection accepted
I (32468) SocketDispatcher: Active sockets: 2
I (33093) Socket: [, 0, 55, 0x3ffdd67c]: Underlying socket closed (recv returned 0)
I (33094) Socket: [, 0, 55, 0x3ffdd67c]: Socket stopping
I (33098) Socket: [, 0, -1, 0x3ffdd67c]: Disconnected
I (33100) SocketDispatcher: Active sockets: 1
```

Am I doing something wrong or misinterpreting the purpose of this test or is the test program meant to behave like this?

PerMalmberg, 2019-10-24:
server_socket_test sets up a TCP/IP server socket and when a connection is accepted, prints the incoming data data as-is (see StreamingClient and StreamingProtocol). There's a note in server_socket_test.cpp about how you can test run it.

You say "datagram", that's TCP/UDP, i.e. a connection less socket. I have not implemented support for that type.

squonk11, 2019-10-24:
Sorry for the misunderstanding, I am not sending UDP datagrams I am sending a TCP/IP telegram. 
If I open the connection using a browser (here: 192.168.1.31:8080) again I only see that a connection is being opened but I don't see any logging of data transmitted. This is my log when I connect using a browser:
```
I (3699867) ServerSocket: Connection accepted
I (3699869) SocketDispatcher: Active sockets: 2
W (3701722) SocketDispatcher: Receive timeout on socket 0x3ffb7bc8 (1500 ms)
I (3701723) Socket: [, 0, 55, 0x3ffb7bc8]: Receive timeout
I (3701723) Socket: [, 0, 55, 0x3ffb7bc8]: Socket stopping
I (3701738) Socket: [, 0, -1, 0x3ffb7bc8]: Disconnected
I (3701739) SocketDispatcher: Active sockets: 1
```
Maybe there is some misunderstanding on my side how the example should work?

PerMalmberg, 2019-10-24:
Hm, I'd expect that too. Unfortunately the CI I have can't actually run the tests, only compile for the different platforms so something might have broken without me noticing.

PerMalmberg, 2019-10-24:
The data is logged on debug-level, enable that in menuconfig and it works. :)

squonk11, 2019-10-24:
yes, in debug mode logging works. I was expecting that "smooth::core::logging::Log" works in the same way as "ESP_LOGI" which also logs in release mode.
Sorry for the stupid question... :-(

PerMalmberg, 2019-10-25:
It does work the same? The data is logged by debug() so unless it is
enabled, nothing is printed.

(when run on the host log levels are ignored and everything is logged)

squonk11, 2019-10-25:
I think the logging issue is now clear for me - so, it can be closed. 
I am currently trying to modify this example for reading bigger data chunks rather than always just one byte. In case I am struggling with that I might open a separate question for this.

squonk11, 2019-11-19:
I am now able to receive incoming data packets, analyze them and cut my protocol frame out of the data stream. I also found out how to send packets from within the StreamingClient class e.g. by: `auto& tx = container->get_tx_buffer();` and `tx.put(p);`. 
Now I would like to send data packets coming from another task (a uart reader in this case). How can I do that? According to my understanding I would need to have access to the instance of the StreamingClient. How can I obtain a reference to it?

PerMalmberg, 2019-11-19:
You should look into queues and share one between the two tasks. Have a look at the task_event_queue in the test folder.

squonk11, 2019-11-20:
I was studying the task_event_queue example. The use of templates and especially the use of nested templates still causes me some headache. But I understood that in this example the `ElapsedTime` is the object being passed from the `Sender Task` to the `App` task, right? 
So, for my application I would have to use the data packet as the object being sent, right?
But what if I have several clients being connected via TCP/IP accessing my resource (the uart in this case). How can I distribute the responses to the corresponding TCP/IP clients?

PerMalmberg, 2019-11-20:
Yes that is correct. Anything that can be copied, like `ElapstedTime` can be put into the queue.

If you have one producer (the UART) and multiple consumer (TCP/IP client) you can use the publiser/subscriber pattern. Smooth has a type-based classes for this: `ipc::Publisher` and `SubscribingTaskEventQueue` that will do all the plumbing for you. See the publish example.

squonk11, 2019-11-20:
in this case what should the `ItemToPublish` class contain? Just the data packet?
I think I need to explain my intended application better:
I have several clients being connected via TCP/IP to the ESP32 and they send requests to the device connected to the uart. So, I am planning to implement some kind of dispatcher task in front of the uart which directs the requests to the device behind the uart and it then should send the response (from the uart) back to the requesting TCP/IP-client. 
Do I have to implement two of such `SubscribingTaskEventQueue`s? One for the communication channel from the TCP/IP-client towards the uart and one for the communication channel back from the uart towards the TCP/IP-client?
Or is there a more clever way to solve this?

PerMalmberg, 2019-11-20:
> in this case what should the ItemToPublish class contain? Just the data packet?

Pretty much, yes.

I assume your clients will be created/destroyed throughout the lifetime of the application so you need a mechanism that can handle that. I assume you will have a fixed number of clients.

The simplest, but not the most effective ways is to use the publisher and attach an ID to the data so each side knows who the recipient is and can discard data meant for others.

The more efficient, but a bit more complicated way is to have all the client send an object containing the data and a shared_ptr to the response queue. This way your dispatcher will know where to queue the data without having to look it up.

A third way is to have the dispatcher create the clients, giving each an ID and a shared response queue on which the clients send an object with data and ID.

HTH.

squonk11, 2019-11-20:
I think solution three is not possible for me because the clients are `ServerClient`s like in the task_event_queue example. So, the clients are created there as soon as a client connects and these are not a fixed number (but limited to a few only; mostly only one). 
Solution two seems to be interesting. The response queue referenced by shared_ptr also has to be a `SubscribingTaskEventQueue`? I guess, yes?
What about the following solution: the data packet passed via a `SubscribingTaskEventQueue` to the dispatcher task could also contain a `this` pointer to the `ServerClient`. Then the dispatcher could call a function of the `ServerClient` class. Then always the right client would be addressed directly, right? Could this also be a solution?

PerMalmberg, 2019-11-20:
IIRC, I made it so that each ServerClient is instantiated in such a way that you can provide an interface to it which would give you way to retrieve the queue.

> The response queue referenced by shared_ptr also has to be a SubscribingTaskEventQueue

No, the `SubscribingTaskEventQueue` is for use with the Publisher; it does some plumbing to tie itself to the Publisher based on the type of data it is templated for. What you want is the `TaskEventQueue`

> dispatcher task could also contain a this pointer to the ServerClient.

I'll answer that question with a question: What will happen if the client disconnects before or while you use the pointer? (Hint: the answer is the reason queues are now held by shared_ptr (they didn't use to be, they were not initially designed to ever be destructed). The destructing_event_queues example test this specific case)

squonk11, 2019-11-23:
In the meantime I got the communication from the `ServerClient` to my dispatcher running using the `Publisher` approach. But here it would be more helpful not to invoke an event but instead to read from the queue just by polling (because I can sent the data to the uart only if there is no other response pending). Is it also possible to read from a queue during a task::tick() when the uart is free?

squonk11, 2019-11-23:
I am currently trying to implement the communication channel from the uart back to the ServerClient. This I am trying to implement as you suggested by using the `TaskEventQueue`. But I am running in a compilation error when I try to attach the initialization of the queue via the constructor. This is my code here:
```    
explicit StreamingClient(smooth::core::Task& task, smooth::core::network::ClientPool<StreamingClient>& pool)
                    : ServerClient<StreamingClient, StreamingProtocol, void>(task, pool, std::make_unique<StreamingProtocol>()),
                      queue(ResponseQueue::create(10, *this, *this))
            {
            }
```
Here I get the error message from the compiler that the 2nd argument to `ResponseQueue::create` cannot be converted: 
```
no known conversion for argument 2 from 'UABus_server_socket::StreamingClient' to 'smooth::core::Task&'
```
I guess that is because StreamingClient is not a task. What can I do in order to solve this?

PerMalmberg, 2019-11-24:
>  Is it also possible to read from a queue during a task::tick() when the uart is free?

The event queues are not meant to be polled, so no. Enqueue the data locally.

> Here I get the error message from the compiler that the 2nd argument to ResponseQueue::create cannot be converted

Each `TaskEventQueue` or `SubscribingTaskEventQueue` must know it's owning task so that it can be ensured that the correct task is running the receiving code. So instead of *this, give it your task.

squonk11, 2019-11-24:
> So instead of *this, give it your task.

Clear so far - but what is the task of the `StreamingClient` class in the `server_socket_test` test?

PerMalmberg, 2019-11-24:
In the case of `server_socket_test`, it's the main task, passed as *this.

https://github.com/PerMalmberg/Smooth/blob/master/test/server_socket_test/server_socket_test.cpp#L55

squonk11, 2019-12-10:
After pausing this project for some time I now continued and implemented a first version working with a `SubscribingTaskEventQueue` for sending data from TCP/IP communication to the uart task. 
But now I would like to send from more than one TCP/IP connection to the uart task. As you mentioned: 

> The event queues are not meant to be polled, so no. Enqueue the data locally.

Is it thread-safe if I write to a queue in the event handler and read from the queue in the tick() function of the same task? I gues yes?

PerMalmberg, 2019-12-11:
Yes. You can use a regular dequeue for that. Just be sure it doesn't grow
unchecked.

squonk11, 2019-12-11:
Is there a way to obtain inside the `tick()` of a `Task` the cycletime of the tick? I would like to use this for managing a timeout by counting tick cycles and the timeout should not change when I modify the tick period in another SW release.

PerMalmberg, 2019-12-11:
I think the tick is exposed in the base class. Otherwise just save it.

I think you're better off to actually track the time instead of number of ticks, there are classes for that in Smooth too.

closed 2020-01-02 by PerMalmberg