PerMalmberg/Smooth #112: Question: Is using UDP communication parallel to Smooth framework possible?

squonk11, 2019-11-02:
Hello,
I have another question: currently I am trying to implement a UDP communication (similar to [this)](https://github.com/espressif/esp-idf/blob/master/examples/protocols/sockets/udp_server/main/udp_server.c) in parallel to the running Smooth framework (with using TCP/IP communication). Unfortunately all calls to `recvfrom` (or similar) are blocking even if data is sent to the listening port. Do you know if this might be related to the TCP/IP communication of Smooth working in parallel? If yes, do you have a hint how I can solve this conflict?

PerMalmberg, 2019-11-02:
`SocketDispacher` and `Socket` do not affect sockets created elsewhere.

If your calls to `recvfrom` blocks, then make sure you've set the sockets to non-blocking. The IDF-exampels are usually using blocking sockets because that's much easier (but less real-world applicable).

squonk11, 2019-11-02:
Thank you for the useful hint. Setting the proper socketoptions did the trick.
This thread can be closed.

closed 2019-11-02 by squonk11