PerMalmberg/Smooth #113: modifications to Wifi.cpp

squonk11, 2019-11-04:
Hello,
I found a bug in Wifi.cpp which results from my implementation of the softap mode: the function `get_mac_address()` always returns the MAC address of the station interface only. But in case softap mode it must return the MAC address of the ap-interface (which actually is different from the MAC address of the station interface).
Shall I fix this and send a PR?
If yes: for my application I would need to get the MAC address and the IP address in binary format (e.g.`uint8_t ip[4]` and `uint8_t mac[6]`) from the Wifi class. Can I implement this also in the same PR?

PerMalmberg, 2019-11-04:
Please, go ahead.

PerMalmberg, 2019-11-05:
I'll close this as #114 in progress.

closed 2019-11-05 by PerMalmberg