PerMalmberg/Smooth #120: Build error on mac for esp platform

flamedmg, 2019-12-22:
[build.log](https://github.com/PerMalmberg/Smooth/files/3993021/build.log)
I tool native_ota_example from esp-idf framework and tried to integrate smooth into it. I failed to compile app then using `idf.py build`. It was building well before.
Please advice. Build log attached.

squonk11, 2019-12-23:
I was facing the same problem after I updated the compiler and the esp-idf version. I temporarily fixed it by removing the compiler options "-Werror", "-Wconversion" and "-Wsign-conversion" from `smooth/lib/compiler_options.cmake.` 
Obviously these options are applied to the complete source (including esp-idf). Maybe the maintainer can find a way how to apply these to the local source of the user only?

PerMalmberg, 2019-12-23:
You're probably building against a newer IDF-version than that Smooth has been tested against, or you're including parts of IDF that has not previously been used together with Smooth.

PerMalmberg, 2019-12-23:
I've just updated the CI pipeline to build using today's master branch and all test projects builds fine locally. What IDF version are you using?

squonk11, 2019-12-23:
I am currently using: `IDF Version:v4.1-dev-474-g2e6398aff-dirty`. But I also newly installed development environment (on another PC) and I guess there is now another version of the compiler also.

PerMalmberg, 2019-12-23:
I built using the one installed by the installer script in IDF.

flamedmg, 2019-12-23:
ESP-IDF v4.1-dev-1086-g93a8603c5
But getting same errors on latest version installed by IDF script

flamedmg, 2019-12-23:
> I was facing the same problem after I updated the compiler and the esp-idf version. I temporarily fixed it by removing the compiler options "-Werror", "-Wconversion" and "-Wsign-conversion" from [...]

I excludede those compiler options and now getting different error, complier complaining on `catch`
```
error: exception handling disabled, use -fexceptions to enable
                 catch (...)
```
Here is a build log: 
[build.log](https://github.com/PerMalmberg/Smooth/files/3994634/build.log)

PerMalmberg, 2019-12-23:
> error: exception handling disabled, use -fexceptions to enable
>                  catch (...)

So enable exceptions in menu config.

flamedmg, 2019-12-28:
Project is compiling now, but i'm getting warning and errors during link:
```
/Users/dmg/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/smooth_component/libsmooth_component.a(Wifi.cpp.obj):(.literal._ZN6smooth4core7network4WifiC2Ev+0x8): undefined reference to `tcpip_adapter_init()'
/Users/dmg/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/smooth_component/libsmooth_component.a(Wifi.cpp.obj):(.literal._ZN6smooth4core7network4Wifi19wifi_event_callbackEPvPKciS3_+0x2c): undefined reference to `tcpip_adapter_set_hostname(tcpip_adapter_if_t, char const*)'
/Users/dmg/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/smooth_component/libsmooth_component.a(Wifi.cpp.obj): in function `smooth::core::network::Wifi::Wifi()':
/Users/dmg/Projects/home-auto/stove/externals/smooth/lib/smooth/core/network/Wifi.cpp:44: undefined reference to `tcpip_adapter_init()'
```

PerMalmberg, 2019-12-28:
What have you tried to fix it?

Expectation management: Smooth is provided free of charge; that doesn't mean that I will fix all your issues. I'll more than likely work with you to to fix any actual issues with the framework or review PRs, but other than that I don't do free consulting.

That said, you probably haven't enabled something in menuconfig so the required code isn't being compiled. There's a sdkonfig in Smooth you can use as a template, or compare against.

closed 2020-01-02 by PerMalmberg