PerMalmberg/Smooth #122: Contributing to smooth.

UtkarshBhatthere, 2020-01-02:
Hey, I am sorry I know this isn't the best way to inquire about contributing.
I've worked a lot in ESP-IDF developing applications for various custom devices and have always felt the need for a better cpp approach for the same. (plus Star for your project) up until now I used to create wrappers around usual API's to make app level development fluent. But this is something way more serious, therefore I seriously wish to contribute to this project.
if there's a current requirements for helping hands (:D) can we discuss how I may proceed with ?

PerMalmberg, 2020-01-02:
Hello @UtkarshBhatthere 

> Hey, I am sorry I know this isn't the best way to inquire about contributing.

Not sure how else you would reach me?

Anyhow, there are a few open issues that you're free to work on, other than that there are nothing else going on. Personally I'm busy with other things but I'll gladly discuss improvements and PRs.

UtkarshBhatthere, 2020-01-03:
Oh great, I'll start looking at the Issues then, I've forked the repo. Shall I let this issue stay ? For people who may wish to inquire the same ?

PerMalmberg, 2020-01-03:
If they want to I'm sure they will get in touch.

closed 2020-01-03 by PerMalmberg