PerMalmberg/Smooth #125: Can this be used for production?

toonst, 2020-01-31:
Hi @PerMalmberg ,

I'm working for a company who is using C++ on the ESP32 in their product. We came across this Smooth library and after some investigation we are really impressed by the elegance and cleanliness of you library. We are considering to use it in a production environment and in general  we're wondering on your thoughts of the readiness of this library.

There's some questions that we had that were not immediately clear from investigating the code:
1. Can `smooth::core::Task` be used for tasks that need to block for a long time? eg. a task waiting for user input, which only runs whenever an event is received from ISR. 

2. `Task::exec()` periodically calls the provided `tick()` function, but what if we want to let our tick function block for a longer time than the tick_interval? Does it still work properly?

Thanks in advance!

PerMalmberg, 2020-01-31:
Hello @toonst ,

I consider it stable enough for [my purposes](https://github.com/PerMalmberg/IO-Card-G3). I've had it running since November-19 without issues. Ultimately it is up to you to judge if it is good enough.

Contributor @enelson1001 might be able to give a 3rd-party opinion on the stability.

1. It's better to set a long-ish tick interval and send an event to the task when the ISR runs, see the [hardware interrupt queue test case/example](https://github.com/PerMalmberg/Smooth/tree/master/test/hw_interrupt_queue).

2. No. [This line](https://github.com/PerMalmberg/Smooth/blob/master/lib/smooth/core/Task.cpp#L148) tries to ensure that the tick is called on schedule but if you block the tick, you also disrupt the tick interval. the result will be that tick() is called immediately after you release the block.

Smooth is designed to be event-driven, i.e. no blocking calls. I've designed it so that this is true for both I/O and network. You're ofc free to do lock up a task if you want, just don't enable the watchdog to restart the device if you do.

PerMalmberg, 2020-02-06:
@toonst Did my reply answer your question?

toonst, 2020-02-07:
Thanks a lot for the explanation! We decided to not start using Smooth yet, since we're on ESP-IDF v3.3 and we have a tight deadline to finish the current product. It seems like the transition from V3.3 to V4 will require some effort. In a next product and once ESP-IDF has made a new release, we're likely going to transition to V4 and start integrating Smooth.

PerMalmberg, 2020-02-07:
Alright, see you then :)

closed 2020-02-07 by PerMalmberg