PerMalmberg/Smooth #130: Rename DisplaySpi to something else

PerMalmberg, 2020-04-26:
`DisplaySpi` is a too generic name. Although it is currently used for three different LCD drivers (IL9341, SH1107 and ST7735) the implementation does not match up with a driver for a WaveShare device which also uses the SPI bus.

@enelson1001 Dou you have a suggestion on a name that reflects the family of devices that the `DisplaySpi` targets?

enelson1001, 2020-04-26:
Maybe LcdSpi, since all three displays are lcd displays.  Similar to what is written in this article 
https://www.r-tt.com/technology-articles/eink-vs-lcd.html

Another option is ActiveSpi (for LCD displays) and StaticSpi (for E-Paper displays) but I don't like as much.

PerMalmberg, 2020-04-26:
Ok, I'll go with simply `LCD` since "display" is part of the acronym and no need to put the bus type in the class name imho. Thanks for your input.

enelson1001, 2020-04-26:
There are some i2c displays also - typically monochrome

PerMalmberg, 2020-04-26:
Yeah, that's true. LCDSpi it is then.

closed 2020-04-26 by PerMalmberg