PerMalmberg/Smooth #134: DS3231 integration

joachimBurket, 2020-09-25:
Hi, 
I'm using your framework in a personal project in which I use a DS3231 RTC module. I have started to write the driver inspiring myself from your PCF8563 code. 
Before I send a pull request, I was wondering how I could regroup the two RTC devices together. For now, they both are under `smooth::application::sensor` namespace. I thought about making a parent class `RtcDevice` regrouping some virtual functions (`get_rtc_time()` and `set_rtc_time()`) and also some utils functions common to the RTC devices (like `bcd_to_decimal()`, `validate_time()` etc..). 

Here is a small UML of the idea: 
[image]

As I am a bit new to OOP principles, I'll be glad to hear about your opinion on this.

PerMalmberg, 2020-09-26:
> I'm using your framework in a personal project

Cool, hope it is enjoyable :)

>  I'll be glad to hear about your opinion on this.

Maybe this?

```
I2CMasterDevice <- RtcDevice <-- DS3231
                           ^--- PCF8563
```

joachimBurket, 2020-10-03:
> Cool, hope it is enjoyable :)
I really  like the design, it's been a long time I was missing a C++ framework to develop on the esp32.

> Maybe this?
The reason I hadn't put the RtcDevice under the I2CMasterDevice was that I thought Rtc could also be in SPI ? 

> There's an old, abandoned PR for the DS3231 you might take inspiration from.
I'll look at it thanks !

PerMalmberg, 2020-10-04:
> The reason I hadn't put the RtcDevice under the I2CMasterDevice was that I thought Rtc could also be in SPI ?

Good point. Maybe an abstract base class then as they'll likely not share any implementation.

PerMalmberg, 2020-11-28:
No activity for a month - closing.

closed 2020-11-28 by PerMalmberg