PerMalmberg/Smooth #135: Using Smooth with stable ESP-IDF versions not possible: not usable for big projects

peterus, 2020-10-13:
Just a quick comment: I wanted to use Smooth with the latest ESP-IDF stable (4.1) version via PlatformIO for a big project.
In PlatformIO I can't switch to the latest (dev/unstable) version of ESP-IDF as the version system is generated from PlatformIO. As I am developing also for beginners which are already overwhelmed with PlatformIO it is very hard to let them run ESP-IDF directly.
So in the end i have to think what I will do now as I really like Smooth but on the other hand it is very hard to use it with stable versions.

Maybe you can add in the future tags or branches which will grow with the official ESP-IDF stable branches - Thanks!

PerMalmberg, 2020-10-14:
@peterus What's the problem with using it with a stable release? Both v4.1 and v4.2 of esp-idf is newer than the latest master that I tried to run Smooth against so unless Espressif made a breaking change then I'd expect it to work.

peterus, 2020-10-14:
@PerMalmberg sorry, of course I forgot to write where it is not working - was already late yesterday :)

the ESP Event system changed from v4.1 to v4.2:
https://github.com/espressif/esp-idf/blob/release/v4.1/components/esp_event/include/esp_event_base.h
https://github.com/espressif/esp-idf/blob/release/v4.2/components/esp_event/include/esp_event_base.h

Now you will get a compile error because of missing `esp_event_handler_instance_t` and some functions.

I am currently not with my development machine, I can post you the exact issue later.

elcojacobs, 2020-10-14:
Just started with smooth and ran into this too. 
I can confirm 4.2 works with no issues. Had the same issue with 4.1.
To update run a `git checkout release/v4.2 --recurse-submodules` in esp-idf dir

For a full reinstall of the tools:
`rm -rf ~/.espressif`
`bash install.sh`

Also run a `idf.py fullclean` in the project

peterus, 2020-10-14:
thanks for the comment @elcojacobs, but as I want to use PlatformIO (not special for me, just for the people which are not used to do a lot on the console) it is a little bit tricky to get 4.2 in PlatformIO as 4.2 is not currently not supported on PlatformIO.

@PerMalmberg here is the error message i got:
```
In file included from ../externals/smooth/lib/smooth/include/smooth/core/Application.h:27,
                 from ../externals/smooth/lib/smooth/core/Application.cpp:26:
../externals/smooth/lib/smooth/include/smooth/core/network/Wifi.h:105:13: error: 'esp_event_handler_instance_t' does not name a type; did you mean 'esp_event_handler_unregister'?
             esp_event_handler_instance_t instance_wifi_event{};
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
             esp_event_handler_unregister
../externals/smooth/lib/smooth/include/smooth/core/network/Wifi.h:106:13: error: 'esp_event_handler_instance_t' does not name a type; did you mean 'esp_event_handler_unregister'?
```

this is just the first issue which the compiler is showing.

PerMalmberg, 2020-10-14:
> Just started with smooth and ran into this too.
> I can confirm 4.2 works with no issues. Had the same issue with 4.1.

Hey, another user. Awesome, @elcojacobs 👍  Please let me know what you think about it, good or bad.

> thanks for the comment @elcojacobs, but as I want to use PlatformIO (not special for me, just for the people which are not used to do a lot on the console) it is a little bit tricky to get 4.2 in PlatformIO as 4.2 is not currently not supported on PlatformIO.
...
> ../externals/smooth/lib/smooth/include/smooth/core/network/Wifi.h:105:13: error: 'esp_event_handler_instance_t' does not 

That looks very familiar. What version of ESP-IDF does PlatformIO support? If I remember correctly I ran Smooth against the master that later became 4.0 and I started using the new event system in PR https://github.com/PerMalmberg/Smooth/pull/80

peterus, 2020-10-14:
@PerMalmberg in PlatformIo you can select all stable versions. Like 4.0 and 4.1.
4.2 is not supported as it is still in development.

PerMalmberg, 2020-10-14:
Hm, did Espressif really not include the new event system in 4.1?!

peterus, 2020-10-14:
@PerMalmberg I was taking a look at the releases from ESP-IDF (luckily Espressif is doing very well documentation) and found out that they introduced the new event system in 4.0.
In 4.1 they added it to the Wifi Lib.

So it looks like they changed the interface from 4.1 to 4.2.

I will take a look if I can somehow create a workaround if this is okay for you.

elcojacobs, 2020-10-14:
> Hey, another user. Awesome, @elcojacobs  Please let me know what you think about it, good or bad.

Looks promising, but still have a lot to figure out how to best restructure my application.
I have added you on Skype, if I have the right person.

PerMalmberg, 2020-10-14:
I haven't logged in on Skype in years...

elcojacobs, 2020-10-14:
What alternative would work better? I use slack, Whatsapp, telegram...

PerMalmberg, 2020-10-15:
> @PerMalmberg in PlatformIo you can select all stable versions. Like 4.0 and 4.1.
> 4.2 is not supported as it is still in development.

Espressif continuously backports/patches stuff to existing releases, at the time of this writing it was only [5 days](https://github.com/espressif/esp-idf/commits/release/v4.1) since a commit was made to 4.1 and [29 days](https://github.com/espressif/esp-idf/commits/release/v4.2) since a commit to 4.2.

> I will take a look if I can somehow create a workaround if this is okay for you.

Not sure what you can do; other than rolling back the commit and whatever else depends on the new event system. I'd push for getting v4.2 into PlatformIO instead as a mod has [already acknowledged that 4.2 is out the door](https://community.platformio.org/t/support-for-esp-idf-v4/11986/2).

peterus, 2020-10-15:
As i was already writing: PlatformIO does support stable releases. 4.2 is still in development and is not a stable release.

Here you can read about it: https://community.platformio.org/t/how-to-use-esp-idf-current-version-from-https-github-com-espressif-esp-idf/4598

I did try yesterday some things, but it is very hard to get this all together and it is not practicable.

From the thread you can also read about an argument: platform_packages
It is not working for 4.2 now, but I will open a ticket on PlatformIO to get this fixed, it is the best way to get around this.

PerMalmberg, 2020-10-15:
Ok seems 4.2 is still in beta.

Anyway, I prefer to keep Smooth working with the master branch of ESP-IDF
because it receives the latest fixes quickest and has in my experience been
the most stable one. Doing this has allowed me to keep up with all
improvements in small steps instead of having to do it all in one go.

This strategy has served me well so far and I see no reason to do otherwise
unless there's something in it for me, or there is a valid reason to stop
using the master branch, which there isn't afaik.

The real problem for you is that Espressif takes such a long time to mark
something as stable. The event system went into the code a year ago and is
apparently still not officially released as a stable release if I read what
was written above correctly.

Im very glad to see all the hard work I put into Smooth is benefitting more
than just myself and I'm happy to review PRs, but I'm not going to let it
lag behind IDF just because some third party won't support the latest
version of IDF.

peterus, 2020-10-15:
@PerMalmberg I will close this Ticket as it is really just an issue with PlatformIO which can be just solved with a new release from Espressif and PlatformIO directly. I did try different different approaches to fix this version hell, but I did not won the fight.

But never the less, I will look for an different approach like also using the master branch from ESP-IDF to do my development and I will think of a simple solution to get the binaries to the not-so-commandline-people :)

Thanks for your work, the library is great and if nobody else will add Ethernet support, I will do it in the next days as I will need it ;)

PS: My current project is a [LoRa APRS iGate and Digipeater](https://github.com/lora-aprs/LoRa_APRS_iGate) for Ham Radio enthusiasts. I will transfer this Arduino project to IDF-ESP with as much C++ as I can get :)

closed 2020-10-15 by peterus

elcojacobs, 2020-10-19:
> Thanks for your work, the library is great and if nobody else will add Ethernet support, I will do it in the next days as I will need it ;)

I have just finished a pull request for that.