PerMalmberg/Smooth #137: Wifi.h:106:13: error: 'esp_event_handler_instance_t' does not name a type

leon-v, 2020-11-28:
Hi !
This will probably be more user error,
But I had installed esp-idf 4.1. I coped the hello world example and updated CMakeLists.txt files and main.cpp and added main.h.
I also updates freertos to 1khz and enables c++ exceptions.
When i compile the basic app i get this error.

Any pointers to lead me in the right direction?

```
FAILED: esp-idf/smooth_component/CMakeFiles/__idf_smooth_component.dir/__/lib/smooth/core/Application.cpp.obj
/home/leon/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++  -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I../externals/smooth/lib/smooth/include -I/home/leon/esp/esp-idf/components -I../externals/smooth/externals/nlohmann_json/include -I../externals/smooth/externals/fmt/include -I/home/leon/esp/esp-idf/components/newlib/platform_include -I/home/leon/esp/esp-idf/components/freertos/include -I/home/leon/esp/esp-idf/components/heap/include -I/home/leon/esp/esp-idf/components/log/include -I/home/leon/esp/esp-idf/components/lwip/include/apps -I/home/leon/esp/esp-idf/components/lwip/include/apps/sntp -I/home/leon/esp/esp-idf/components/lwip/lwip/src/include -I/home/leon/esp/esp-idf/components/lwip/port/esp32/include -I/home/leon/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/leon/esp/esp-idf/components/soc/esp32/include -I/home/leon/esp/esp-idf/components/soc/include -I/home/leon/esp/esp-idf/components/esp_rom/include -I/home/leon/esp/esp-idf/components/esp_common/include -I/home/leon/esp/esp-idf/components/xtensa/include -I/home/leon/esp/esp-idf/components/xtensa/esp32/include -I/home/leon/esp/esp-idf/components/esp32/include -I/home/leon/esp/esp-idf/components/driver/include -I/home/leon/esp/esp-idf/components/driver/esp32/include -I/home/leon/esp/esp-idf/components/esp_ringbuf/include -I/home/leon/esp/esp-idf/components/efuse/include -I/home/leon/esp/esp-idf/components/efuse/esp32/include -I/home/leon/esp/esp-idf/components/vfs/include -I/home/leon/esp/esp-idf/components/esp_wifi/include -I/home/leon/esp/esp-idf/components/esp_wifi/esp32/include -I/home/leon/esp/esp-idf/components/esp_event/include -I/home/leon/esp/esp-idf/components/esp_netif/include -I/home/leon/esp/esp-idf/components/esp_eth/include -I/home/leon/esp/esp-idf/components/tcpip_adapter/include -I/home/leon/esp/esp-idf/components/app_trace/include -I/home/leon/esp/esp-idf/components/mbedtls/port/include -I/home/leon/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/leon/esp/esp-idf/components/json/cJSON -I/home/leon/esp/esp-idf/components/pthread/include -I/home/leon/esp/esp-idf/components/nvs_flash/include -I/home/leon/esp/esp-idf/components/spi_flash/include -I/home/leon/esp/esp-idf/components/fatfs/diskio -I/home/leon/esp/esp-idf/components/fatfs/vfs -I/home/leon/esp/esp-idf/components/fatfs/src -I/home/leon/esp/esp-idf/components/wear_levelling/include -I/home/leon/esp/esp-idf/components/sdmmc/include -I/home/leon/esp/esp-idf/components/libsodium/libsodium/src/libsodium/include -I/home/leon/esp/esp-idf/components/libsodium/port_include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu++11 -fexceptions -fno-rtti -D_GNU_SOURCE -DIDF_VER=\"v4.1\" -DESP_PLATFORM -Wnon-virtual-dtor -Wold-style-cast -Werror -Wcast-align -Wunused -Wconversion -Wsign-conversion -Wlogical-op -Wdouble-promotion -Wformat=2 -Wimplicit-fallthrough -Wmisleading-indentation -Wduplicated-cond -Wduplicated-branches -Wnull-dereference -std=gnu++17 -MD -MT esp-idf/smooth_component/CMakeFiles/__idf_smooth_component.dir/__/lib/smooth/core/Application.cpp.obj -MF esp-idf/smooth_component/CMakeFiles/__idf_smooth_component.dir/__/lib/smooth/core/Application.cpp.obj.d -o esp-idf/smooth_component/CMakeFiles/__idf_smooth_component.dir/__/lib/smooth/core/Application.cpp.obj -c ../externals/smooth/lib/smooth/core/Application.cpp
In file included from ../externals/smooth/lib/smooth/include/smooth/core/Application.h:27,
                 from ../externals/smooth/lib/smooth/core/Application.cpp:26:
../externals/smooth/lib/smooth/include/smooth/core/network/Wifi.h:105:13: error: 'esp_event_handler_instance_t' does not name a type; did you mean 'esp_event_handler_unregister'?
             esp_event_handler_instance_t instance_wifi_event{};
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
             esp_event_handler_unregister
../externals/smooth/lib/smooth/include/smooth/core/network/Wifi.h:106:13: error: 'esp_event_handler_instance_t' does not name a type; did you mean 'esp_event_handler_unregister'?
             esp_event_handler_instance_t instance_ip_event{};
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
             esp_event_handler_unregister
[803/835] Building CXX object esp-idf/smooth_component/CMakeFiles/__idf_smooth_component.dir/__/lib/smooth/core/filesystem/SPIFlash.cpp.obj
ninja: build stopped: subcommand failed.
```

PerMalmberg, 2020-11-28:
For whatever reason v4.1[ does not contain](https://github.com/espressif/esp-idf/blob/release/v4.1/components/esp_event/include/esp_event_base.h#L27-L32) the `esp_event_handler_instance_t` definition.

[v4.2 does](https://github.com/espressif/esp-idf/blob/release/v4.2/components/esp_event/include/esp_event_base.h#L33), so your only option is to update to 4.2, or another appropriate branch if IDF

leon-v, 2020-11-28:
Thanks for the prompt reply.
No idea why, but they moved it [private.](https://github.com/espressif/esp-idf/blob/release/v4.1/components/esp_event/private_include/esp_event_internal.h#L36)

I will switch to 4.2.

Thanks again!

closed 2020-12-01 by PerMalmberg