PerMalmberg/Smooth #138: Possibility to stop the HTTP server?

squonk11, 2020-12-27:
After working on a Bluetooth project with ESP32 I now come back to WiFi. 
Currently I am trying to implement a HTTP server page for configuration settings which is only available to the enduser if he has physical access to the ESP32 and can press a button to activate this HTTP server. Launching this server is not an issue but after all configurations on the config web page have been done, it should be possible to shut down this HTTP server again. Unfortunately in HTTPServer there is no stop() method. 
Do you have a hint how I can shut down the server (preferred) or at least make the config page invisible?

PerMalmberg, 2020-12-29:
You could run a check against some persistent data to see what you should return in the request handler.

I've never thought about the server needing to be shut down but I think just deleting the instance would do it? Haven't checked though.

squonk11, 2020-12-29:
I think it is better to activate a different page showing a hint how to activate the config page (e.g. by pressing a button). So, the server does not need to be shut down.
But in this case I need to manage two different handlers for the Path "\" and switch between these. Is there an easy way to delete a handler which was previously attached to the server via the HTTPServer.on() method?

squonk11, 2020-12-29:
It seems to work like described in the Edit section of my previous post. So I'll close this.

closed 2020-12-29 by squonk11