PerMalmberg/Smooth #139: style sheet of webpage not respected

squonk11, 2020-12-30:
Currently I am encountering a very strange issue: I copied a sample website from [SelfHTML](https://src.selfhtml.org/) which I copied to my HTTPServer web_root. The page uses just some style sheets from a file in the subdirectory /css. But the resulting page I see in the browser does obviously not use the style sheet. I did the following tests:

- loading the same page (same files) from hard disk works perfectly (so the page seems to be designed correctly)
- When I copy style.css from web_root/css to web_root and modify index.html accordingly I get the same mess (so it is not an issue with reading from a subdirectory)
- I checked the log of Smooth: all files seem to be transmitted correctly from the ESP32 to the browser - all files have the message `Reply: OK`; also the file /css/style.css (so the webserver seems to work o.k.)
- I checked the wireshark log: no error messages. The files /index.html and /css/style.css have the correct contents when being sent. (so the network communication also seems to work)

Now I am somehow lost. Why the heck the webpage works perfectly when being loaded from harddisk and does not show up correctly when being loaded from the webserver?
Do you have any idea what else I could check? Maybe there is a header information missing in the reply?

PerMalmberg, 2020-12-31:
This function needs some expansion:

https://github.com/PerMalmberg/Smooth/blob/0e277276623a89b7484c8d9205776436ca947502/lib/smooth/application/network/http/http_utils.cpp#L87

squonk11, 2020-12-31:
ah, yes. I probably would have found it myself but I had to follow other tasks yesterday after my 2nd edit.
So, I think it might make sense to add some more mime types. A maybe complete list can be found here: [mime-types](http://www.iana.org/assignments/media-types/media-types.xhtml). But implementing all in Smooth would probably be too much. So an extract of the most frequent ones which might possibly be used in combination with an ESP32 application would be meaningful. 
If you don't mind I could select some and add them to a PR.

PerMalmberg, 2021-01-04:
> If you don't mind I could select some and add them to a PR.

No, of course not. Please do.

squonk11, 2021-01-06:
Currently I am again quite busy. I hope I will be able to do it during the weekend. 
Additionally there is again a compilation issue due to an old style cast using the latest master branch. Probably again a `#pragma GCC diagnostic push` is needed. Maybe I can find a fix for this too.

PerMalmberg, 2021-01-12:
I'll close this as the PR is in progress.

closed 2021-01-12 by PerMalmberg