PerMalmberg/Smooth #141: Problems with Wifi.cpp and SDCard.cpp

enelson1001, 2021-02-23:
I hope you are doing well and still maintaining Smooth.

I guess the first question before getting into the issues is which version of esp-idf should be used with Smooth.  The Requirements section on your README.md just states esp-idf-v4.x.

### Wifi.cpp
1. I ran into issues with Wifi.cpp when using the Master branch on esp-idf (I did not have problems when using v4.2) in that Wifi.cpp will not compile.  Based on esp-idf example - examples/wifi/getting_started/softAP/main/softap_example_main.c.  looks like line 297 in Wifi.cpp needs to be changed to the following: `esp_wifi_set_config(WIFI_IF_AP, &config); `
2. I would like to retrieve gateway and mask information using `esp_netif_get_ip_info(esp_netif_t *esp_netif, esp_netif_ip_info_t *ip_info)`  but don't know how to do this with existing implementation.
3. When I look at my wifi router I never see the host named that I assigned in the program. My wifi router always shows unknown. I am using DHCP.  Any suggestions? 

### SDCard.cpp
1. I ran into issues using esp-idf-Master and esp-idf-v4.2 where SDCard.cpp will not compile. The SDCard.ccp looks like the init method needs to be updated to something like the following based on esp-idf example - examples/storage/sd_card/main/sd_card_example_main.c
```
void SDCard::init()
    {
        Log::info(TAG, "Initializing SDCard");
       
        esp_vfs_fat_sdmmc_mount_config_t mount_config{};
        mount_config.format_if_mount_failed = false;
        mount_config.max_files = 5;
        //mount_config.allocation_unit_size = 16 * 1024;

        host = (sdmmc_host_t)SDSPI_HOST_DEFAULT();
        Log::error(TAG, "slot = {}", host.slot);

        host.slot = VSPI_HOST;
        Log::error("SPISDCard", "slot = {}", host.slot);
        
        //bus_cfg.miso_io_num = GPIO_NUM_19,       // MISO
        //bus_cfg.mosi_io_num = GPIO_NUM_23,       // MOSI
        //bus_cfg.sclk_io_num = GPIO_NUM_18,       // CLK
        //bus_cfg.quadwp_io_num = -1;
        //bus_cfg.quadhd_io_num = -1;
        //bus_cfg.max_transfer_sz = 4000;

        //auto res = spi_bus_initialize(static_cast<spi_host_device_t>(host.slot), &bus_cfg, 1);

        //if (res != ESP_OK) 
        //{
            //Log::error(TAG, "Failed to initialize bus.");
        //}

        sdspi_device_config_t slot_config = SDSPI_DEVICE_CONFIG_DEFAULT();
        slot_config.gpio_cs = GPIO_NUM_4;
        slot_config.host_id = static_cast<spi_host_device_t>(host.slot);

        auto mount_result = esp_vfs_fat_sdspi_mount("/sdcard", &host, &slot_config, &mount_config, &card);

        bool initialized = mount_result == ESP_OK;

        if (initialized)
        {
            Log::info(TAG, "SD Card initialized");
            sdmmc_card_print_info(stdout, card);
        }
        else
        {
            if (mount_result == ESP_FAIL)
            {
                Log::error(TAG, "Failed to mount the file system.");
            }
            else
            {
                Log::error(TAG, "Failed to initialize SD Card.");
            }
        }
    } 
```
But the spibus needs to be initialized.  For me I am using the M5Stack which has the display and SDCard on the same spi bus.  I am initializing the spi bus in my DisplayDriver class, so that is why you see the spi_bus_initialize commented out.  Not sure how to handle this use case in the SDCard.cpp.  I did have to implement a 1 sec Thread:Sleep between the DisplayDriverTask.start() and the SDCard.init() but that seemed reasonable to me.
2. I also need to pass into constructor the host.slot to be used as the SDSPI_HOST_DEFAULT() has the slot.host set to HSPI_HOST and I need to use VSPI_HOST.

PerMalmberg, 2021-02-25:
I've always used the master branch of IDF, but since I'm currently spending my free time on things outside the world of microprocessors I've not kept tabs on what has changed and updated accordingly.

1. So Espressif has changed the API in a breaking way? Not surprising as they can do pretty much what they want with the master branch, it is after all not considered stable in terms of API compatibility. Based on previous experience I'd say that whatever goes into master is what will eventually be released as a major version so updating Smooth to work against master is likely the best way.
2. Neither do I without digging deeper.
3. No.... mine shows "espressif" for my device running Smooth on my Wifi. Not sure I've ever seen this work or not.

SDCard: So they have broken the API again? Maybe it'd be better to break out the initialization into a separate object that is passed to the different drivers (sdcard and display) so that it is initialized in a single place?

enelson1001, 2021-02-25:
### Wifi.cpp
1. So I assume you will be updating sometime soon? If not I will use esp-idf-v4.2
2. Would be nice to have method to retrieve AP IP address and a method to retrieve AP mask.
3. Your's shows "espressif" even when you change your host name?

### SDCard.cpp
1. Sounds reasonable. Guess that means updating LCDSpi and BME280Spi correct?

squonk11, 2021-02-27:
concerning Wifi.cpp:

1. yes, the line 297 must be changed to `esp_wifi_set_config(WIFI_IF_AP, &config);`. Otherwise it will not compile in V4.3.
2. for gateway and mask information you can do the following: in `Wifi::wifi_event_callback` for the event `IP_EVENT` starting in line in line 193 you can add `wifi->ip_info = reinterpret_cast<ip_event_got_ip_t *>(event_data)->ip_info;` because the ip_info is included in the event data. Then you can add to Wifi.h:
```            
esp_ip4_addr_t *get_ip() { return &ip_info.ip; };
esp_ip4_addr_t *get_netmask() { return &ip_info.netmask; };
esp_ip4_addr_t *get_gateway() { return &ip_info.gw; };
```
in order to get the needed info.
3. I am setting the AP name using `void set_host_name(const std::string& name);`. After that the router shows the name as expected.

concerning SDCard.cpp: unfortunately I am not so familiar with  SPI.

PerMalmberg, 2021-03-01:
@squonk11 As you can see from my somewhat slow response, my priorities are currently elsewhere. I'd go with IDF 4.2 for now. I'm happy to look at a PR for on update to IDF-master, but I won't be making the updating myself in the near future.

squonk11, 2021-03-01:
I am also quite busy these days but I can try to make a PR as soon as possible.

enelson1001, 2021-03-03:
1. I reset my router and hostname is working.
2. I have update Wifi.cpp and Wifi.h and will submit a PR shortly
3.  Still looking at SDCard and LCDSpi to figure out how to proceed.

PerMalmberg, 2021-03-16:
Closing in favor of #145

closed 2021-03-16 by PerMalmberg