PerMalmberg/Smooth #145: Discussion: SPI Master & SDCard updates

enelson1001, 2021-03-15:
### Background
1. To create an SDCard object with the latest esp-idf requires that the spi-bus-master be initialized using spi_bus_initialize(host, bus_config, dma_chan).  The current implementation of smooth::core::io::spi::Master initializes the spi-bus-master as part of the creation of an SPIDevice.  The smooth::core::io::spi::Master needs to be updated to separate the spi-bus-master initialization from the creation of an SPIDevice.
2. The spi-bus-master must only be initialized once using - spi_bus_initialize(host, bus_config, dma_chan).  If you try and initialize a second time esp-idf will throw an error.
3. In my case I am using the M5Stack which has the LCD and the SDCard or on the same spi-bus (VSPI_HOST).
4. The SDcard class should be modified so the constructor includes which spi_host is being used for the spi-bus-master.  The current implementation assumes the spi-host is hspi_host.  

### Approach #1
In this approach a singleton is used in creating a Master.  The Master has an empty constructor and some new methods.

**Pros:** 
1. Allows the initialization of a hspi-bus-master or vspi-bus-master. 
2. Separates the creation of an SPIDevice from the initialization of spi-bus-master.  
3. Prevents the initializing of a particular spi-bus-master (hspi or vspi) more than once.

**Cons:** 
1. Doesn't guarantee that the spi-bus-mater has been initialized before creating an SPIDevice. We could have two methods for device creation instead of one to eliminate this problem - std::unique_ptr<DeviceType> create_hspi_device(Args&& ... args) and std::unique_ptr<DeviceType> create_vspi_device(Args&& ... args)
2. Singletons seem to frowned upon in the programming community.

**Implementation** details: 
I placed spi-bus-master initialization inside my DisplayDriver.cpp and inside SDCard.cpp.  The spi-bus-master was initialized using the following statement.
```
         Master::get_instance().initialize_vspi_master(SPI_DMA::DMA_BLOCK_1,         // use DMA
                                                         GPIO_NUM_23,                // mosi gpio pin
                                                         GPIO_NUM_19,                // miso gpio pin  (full duplex)
                                                         GPIO_NUM_18,                // clock gpio pin
                                                         MAX_DMA_LEN);               // max transfer size
```

The following is the header file MasterV1.
```
MasterV1.h
#pragma once

#include <memory>
#include <mutex>
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wconversion"
#pragma GCC diagnostic ignored "-Wsign-conversion"
#pragma GCC diagnostic ignored "-Wold-style-cast"
#include <driver/gpio.h>
#include <driver/spi_common.h>
#include <driver/spi_master.h>
#pragma GCC diagnostic pop

namespace smooth::core::io::spi
{
    enum SPI_DMA
    {
        NO_DMA = 0,
        DMA_BLOCK_1,
        DMA_BLOCK_2,
    };

    class MasterV1
    {
        public:
            MasterV1(const MasterV1&) = delete;                     // Copy constructor prohibited  
            MasterV1(MasterV1&&) = delete;                          // Move constructor prohibited
            MasterV1& operator=(const MasterV1&) = delete;          // Assignment operator prohibited
            MasterV1& operator=(MasterV1&&) = delete;               // Move operator prohibited

            static MasterV1& get_instance();

            void initialize_hspi_master(SPI_DMA dma_channel,
                                        gpio_num_t mosi,
                                        gpio_num_t miso,
                                        gpio_num_t clock,
                                        int transfer_size = 0,
                                        gpio_num_t quadwp_io_num = GPIO_NUM_NC,
                                        gpio_num_t quadhd_io_num = GPIO_NUM_NC
[... 390 lines cut ...]
        //if (mount_result)
        {
            Log::info(TAG, "SD Card initialized");
            sdmmc_card_print_info(stdout, card);
        }
        else
        {
            if (mount_result == ESP_FAIL)
            {
                Log::error(TAG, "Failed to mount the file system.");
            }
            else
            {
                Log::error(TAG, "Failed to initialize SD Card.");
            }
        }

        FSLock::set_limit(max_file_count);

        return initialized;
    }
```

### Recommendations
Are any of these approaches a good staring point for a pull request or did you have something compeletey different in mind?. Let me know how you want to proceed.

PerMalmberg, 2021-03-16:
I like the first approach though there's no need for a singleton, I think. Make the members static instead and have the create_-functions be static too. The drawback to that solutions is that you can't destroy the master before program exit, but I don't think that is a problem?

enelson1001, 2021-03-23:
This discussion can be closed

closed 2021-03-23 by PerMalmberg