PerMalmberg/Smooth #147: SystemStatistics.h needs #pragma once

Paul-Simpson, 2021-03-21:
..otherwise compilation errors out with:

In file included from ../components/motor/include/motor.h:7,
                 from ../main/main.h:10,
                 from ../main/main.cpp:8:
../externals/smooth/lib/smooth/include/smooth/core/SystemStatistics.h:55:11: error: redefinition of 'class smooth::core::SystemStatistics'
     class SystemStatistics

Adding the pragma fixes the issue.

PS - thank you for sharing this framework. It's early days for me, but I'm loving it. :)

PerMalmberg, 2021-03-22:
> PS - thank you for sharing this framework. It's early days for me, but I'm loving it. :)

Glad it can be of help.

Please open a PR with the requested change and I'll merge it to master.

Paul-Simpson, 2021-03-22:
Done.

PerMalmberg, 2021-03-23:
Merged.

closed 2021-03-23 by PerMalmberg