PerMalmberg/Smooth #153: use fill capability of cmake

peterus, 2021-04-03:
The full capability of cmake should be used: using ninja and ctest

the use of ctest can improve the github actions runtime as cmake will compile all testcases in parallel and not sequential (it will definitely improve on the local machines with more cpu cores). and ctest is more robust than using a generic bash-script.

mixing platforms (in this case host and esp32 build) should not be done in one build folder. splitting up in a `build_host` and `build_esp` folder would be another small suggestion (than we can also use ninja for the host build which is taking care of parallel builds).

PerMalmberg, 2021-04-03:
Interesting. I've never used ctest myself, but I'm all for improvements. Please, go ahead and show us some ctest-magic :)

PerMalmberg, 2021-05-17:
No activity since beginning of April, closing. Please open a new ticket for ctest if this is still relevant with the latest changes.

closed 2021-05-17 by PerMalmberg