PerMalmberg/Smooth #157: HTTP Server not working on host system

peterus, 2021-04-14:
Due to the changes in docker and finding out which ports I have to open I was testing the `http_server_test`.

Unfortunately it is not working on the host system as `__FILE__` has no relativ path (`../../test/http_server_test/http_server_test.cpp`).
The check in HTTPServer.h (line 234) `config.web_root().is_parent_of(search)` is always false because the function `is_parent_of()` is working just for relative paths.

PerMalmberg, 2021-04-17:
I am a bit short on time to dig, can you please provide some more details on why using docker breaks the HTTP server or am I misunderstanding you?

PerMalmberg, 2021-05-17:
No response, closing. Please reopen if this is still an issue.

closed 2021-05-17 by PerMalmberg