PerMalmberg/Smooth #160: C++ code vs expectations

Reneg973, 2021-04-17:
I checked my first ESP_IDF project and - as a C++ I am not very satisfied. So googled for wrapper around ESP_IDF and found this repository.

I try to use it, but there are things that - even if they are correct from C++ point of view, not really convenient.
One really simple example:

```
Input in10(pin10)
Input in11(pin11);
if(in10 == in11) {} // always returns false atm
```
Of course it is possible to write `in10.read() == in11.read())`, but I would expect the first code to be working too.

Other questions:
- What is your rule about const correctness?
- What is your limit on the number of function parameters? From my pov 13 input parameters are a little bit too much.
- I read your rule to exceptions. My question here is, if you write code which won't work at all, shouldn't it throw an exception?

P.S. in case of `constexpr`, an additional `const` is superfluous.

Reneg973, 2021-04-17:
Additionally I really would use `gpio_pull_mode_t` instead of 2x `bool` parameters for the Input constructor.

PerMalmberg, 2021-04-18:
Sorry to say this, but they way you put forward your opinions is a bit offensive. If you want people to listen to your opinions consider being a bit humble. You did not write this library, nor do you have the knowledge of why certain design decisions were made.

That said, I'm going to take time out of my day to answer your questions.

> I try to use it, but there are things that - even if they are correct from C++ point of view, not really convenient.
> One really simple example:
>
> ```
> Input in10(pin10);
> Input in11(pin11);
> if(in10 == [...]

You're free to have your opinion, but `operator==` is already defined for another purpose. Operator overloading has always been a matter for discussion in the C++ community and there is really no right and wrong, only context-based design decisions to be made. Being able to do `if(input1) { /* pin is active */ }` may be convenient and perhaps I'll make a breaking change to the API sometime if I deem it worth the effort.

>  What is your rule about const correctness?

const is nice, but it also causes some issues if not applied properly. I don't use it often enough. Not everything needs to be const.

> What is your limit on the number of function parameters? From my pov 13 input parameters are a little bit too many.

Depends on the context. 13 is a lot, I presume you found that in Smooth somewhere? The alternative is passing a struct, but then you'd have to populate that instead.

>     * I read your rule to exceptions. My question here is, if you write code which won't work at all, shouldn't it throw an exception?
>       short example:
> 
> 
> ```
> Input in(an_invalid_pin_id);  // exception, hey programmer, use your head!
> while(in.read() != 1) // this code won't do it
> ```
> 
> Any oppositions?

Yes. Just like goto, exceptions is another code path the program can take. And instead of testing for bad input for _each time the code runs_ put some responsibility on the programmer to do things right.

> 
> P.S. in case of `constexpr`, an additional `const` is superfluous.

Example?

> Additionally I really would use `gpio_pull_mode_t` instead of 2x `bool` parameters for the Input constructor.

Again, your opinion. The purpose of Smooth is to abstract away ESP-IDF and as such as little as possible of the types should leak through to the user.

Reneg973, 2021-04-18:
Excuse me, was a little bit late and I was frustrated, that I bricked 2 of my XIAO SAMD devices without being my fault. The upload just stopped with an error and after that no upload possible anymore...

First thank you for your time to give a good and complete answer. I worked long time with mbed, switched to Arduino for some days and now - as I want to connect my mbed to WiFi, checked the ESP-IDF. Unfortunately this IDF is not very open to beginners. I thought using a C++ wrapper could make it easier.
You may know, every API is different, and all have their own design decisions. One of the time consuming things is to find out, why code written with one API doesn't work with another.

> Being able to do` if(input1) { /* pin is active */ }` may be convenient and perhaps I'll make a breaking change

I don't think this is a breaking change,  you may simply add `operator bool() const { return read(); }`. :)

> Not everything needs to be const.

Of course nothing needs to be const. It was just a question, how you would like to see it handled within this API.

> 13 is a lot, I presume you found that in Smooth somewhere? 

Yes found it in SPI constructor.

> P.S. in case of constexpr, an additional const is superfluous.
>> Example?

Forget it, I've overseen the pointer.

> And instead of testing for bad input for each time the code runs put some responsibility on the programmer

I really miss that with ESP-IDF. Almost every function checks for valid parameters. Totally superfluous as in most cases the functions are called with constant predefined pin IDs.

> Additionally I really would use gpio_pull_mode_t instead of 2x bool parameters for the Input constructor.
> > Again, your opinion. The purpose of Smooth is to abstract away ESP-IDF and as such as little as possible of the types should leak through to the user.

I understand that. I just don't understand why doing it for one type but not for others.

PerMalmberg, 2021-04-18:
> Unfortunately this IDF is not very open to beginners. I thought using a C++ wrapper could make it easier.

Imho it's not so bad, though I, like you seem to do, prefer C++ over C which is why I wrote Smooth. At least the there is plenty of documentation for when you have to dig into something.

> I don't think this is a breaking change, you may simply add operator bool() const { return read(); }. :)

I think that will result in unambiguous `operator==` when doing `input1 == input2` as in your example. A simple `if(input)` would work though.

> Yes found it in SPI constructor.

I'm open to improvements if you have them.

> I understand that. I just don't understand why doing it for one type but not for others.

Probably because I didn't have anyone checking my code to remind me of my own goals . :)  And, it is a goal, not an absolute must.

Reneg973, 2021-04-18:
>I think that will result in unambiguous operator== when doing input1 == input2 as in your example.

The result of (input1 == input2) and (bool)input1 == (bool)input2 may still be different.

Just to see whether my kind of thinking would be accepted, I post a small template here. Please check also my comments.

```
class Input
{
public:
  enum class PullMode
  {
    pull_up,
    pull_down,
    pull_both,
    pull_none
  };

  explicit Input(gpio_num_t input_pin);

  Input(gpio_num_t input_pin,
        PullMode pull_mode);

  ~Input() = default; // do you really want to destroy derived classes via base class? I would remove virtual or destructor at all

  bool read() const;

  Input(const Input&) = delete; // why not allowing a copy? It's just a pin which is copied. Disallowing it could confuse C++ newbies but OTOH leads to better readability

  Input& operator=(const Input&) = delete; // same as copy constructor

  bool operator==(const Input& other) const;

  bool operator!=(const Input& other) const // is inline code allowed? If not, why not?
  {  return !operator==(other); }

  operator bool() const // is inline code allowed? If not, why not?
  { return read(); }

protected:
  Input(gpio_num_t input_pin,
        PullMode pull_mode,
        gpio_int_type_t interrupt_type = GPIO_INTR_DISABLE); // reserve interrupt_type for derived class

  const gpio_num_t io;
};
```

Please note, virtual destructors are no problem. Atm I just don't see any use case to have it for this class.

My thoughts of having a more abstract class library go into this direction:
https://os.mbed.com/docs/mbed-os/v6.9/apis/index.html

If you think we could pull  together let me know.

closed 2021-05-17 by PerMalmberg