PerMalmberg/Smooth #162: Trouble running build_test.sh

enelson1001, 2021-07-12:
I am trying to run build_test.sh and am seeing the following message.  Any suggestions?
```
ed@ed-ideacentre-AIO-910-27ISH:~/smooth-forked-repo-two/Smooth/CI$ ./build_test.sh 
Building Host -------------------------------------------------------
WARNING: The ESP_IDF_VERSION variable is not set. Defaulting to a blank string.
ERROR: Version in "./../docker-compose.yml" is unsupported. You might be seeing this error because you're using the wrong Compose file version. Either specify a supported version (e.g "2.2" or "3.3") and place your service definitions under the `services` key, or omit the `version` key and place your service definitions at the root of the file to use version 1.
For more on the Compose file format versions, see https://docs.docker.com/compose/compose-file/
```

enelson1001, 2021-07-12:
Moved up one level in directory but still see the following error.

```
ed@ed-ideacentre-AIO-910-27ISH:~/smooth-forked-repo-two/Smooth$ ./CI/build_test.sh 
Building Host -------------------------------------------------------
ERROR: Version in "./docker-compose.yml" is unsupported. You might be seeing this error because you're using the wrong Compose file version. Either specify a supported version (e.g "2.2" or "3.3") and place your service definitions under the `services` key, or omit the `version` key and place your service definitions at the root of the file to use version 1.
For more on the Compose file format versions, see https://docs.docker.com/compose/compose-file/
```

enelson1001, 2021-07-15:
Was able to fix the above error message by upgrading docker-compose.

The version on my linux PC was 1.25.0
```
$ docker-compose -v
docker-compose version 1.25.0, build unknown
```

Performed the following steps to upgrade on my linux PC
```
$ which docker-compose
/usr/bin/docker-compose

$ sudo rm /usr/bin/docker-compose

$ sudo curl -L https://github.com/docker/compose/releases/download/1.29.2/docker-compose-`uname -s`-`uname -m` -o /usr/bin/docker-compose

$ sudo chmod +x /usr/bin/docker-compose

$ sudo docker-compose -v 
docker-compose version 1.29.2, build 5becea4c
```

enelson1001, 2021-07-15:
Was able to run the following with NO problems
```
$ docker-compose run --rm smooth ./CI/build_smooth_host.sh

$ docker-compose run --rm -w /src/build/host/test/linux_unit_tests smooth ./linux_unit_tests

$ docker-compose run --rm smooth ./CI/build_smooth_esp32.sh
```

But when I try and run 
```
$ ./CI/build_test.sh
```

I get the following error
```
Running host tests ---------------------------------------------------
Creating smooth_smooth_run ... done
Detecting the Python interpreter
Checking "python" ...
Python 3.6.9
"python" has been detected
Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool cmake found in PATH: 3.10.2.
Using Python interpreter in /opt/esp/python_env/idf4.4_py3.6_env/bin/python
Checking if Python packages are up to date...
Python requirements from /opt/esp/idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /opt/esp/idf/components/esptool_py/esptool
  /opt/esp/idf/components/espcoredump
  /opt/esp/idf/components/partition_table
  /opt/esp/idf/components/app_update
  /opt/esp/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin
  /opt/esp/tools/xtensa-esp32s2-elf/esp-2021r1-8.4.0/xtensa-esp32s2-elf/bin
  /opt/esp/tools/xtensa-esp32s3-elf/esp-2021r1-8.4.0/xtensa-esp32s3-elf/bin
  /opt/esp/tools/riscv32-esp-elf/esp-2021r1-8.4.0/riscv32-esp-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin
  /opt/esp/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin
  /opt/esp/tools/cmake/3.16.4/bin
  /opt/esp/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/bin
  /opt/esp/python_env/idf4.4_py3.6_env/bin
  /opt/esp/idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

/opt/esp/entrypoint.sh: line 6: /src/build_host/test/linux_unit_tests/linux_unit_tests: No such file or directory
ERROR: 127
```

enelson1001, 2021-07-15:
I do have the following folder (it's permission are set to root).
```
build_host/test/linux_unit_tests
```

But not folder listed in the error description.
```
build_host/test/linux_unit_tests/linux_unit_tests
```

_Should/will this folder be deleted after script completes?_

enelson1001, 2021-07-15:
OK solved 2nd error.

build_test.sh had errors on line 6 and line 8.

```
ORIGINAL build_test.sh
===================
#!/bin/bash

set -e

echo "Building Host -------------------------------------------------------"
docker-compose run smooth ./CI//build_smooth_host.sh
echo "Running host tests ---------------------------------------------------"
docker-compose run -w /src/build_host/test/linux_unit_tests smooth ./linux_unit_tests
echo "Building esp32 test projects -----------------------------------------"
docker-compose run smooth ./CI/build_smooth_esp32.sh
```

Modified and now looks like.
```
#!/bin/bash

set -e

echo "Building Host -------------------------------------------------------"
docker-compose run smooth ./CI/build_smooth_host.sh
echo "Running host tests ---------------------------------------------------"
docker-compose run -w /src/build/host/test/linux_unit_tests smooth ./linux_unit_tests
echo "Building esp32 test projects -----------------------------------------"
docker-compose run smooth ./CI/build_smooth_esp32.sh
```
I will remove build_host folder since it is not required.

PerMalmberg, 2021-08-07:
I saw you made a change re. build host in the PR, this is all sorted then?

enelson1001, 2021-08-08:
Yes you can close

closed 2021-08-09 by PerMalmberg