PerMalmberg/Smooth #164: ❓Can you require smooth from a traditional ESP-IDF component?

jsmestad, 2021-07-30:
I apologize if this is answered somewhere else, I could not find it 🙁

I am slowly converting a project over to use Smooth (great work by the way!), and I have a component that I have switched over to using `smooth::core::io::Input`, but the compiler cannot find the header.

Directory Structure:
```
components/
|- encoder/
    |- include/
      |- rotary_encoder.hpp
    |- CMakeLists.txt
    |- rotary_encoder.cpp
|- externals/
   |- smooth/..
```

When I add `smooth` to the component's `REQUIRES` definition, it still cannot find the header:

```
# components/encoder/CMakeLists.txt

idf_component_register(
    SRCS "rotary_encoder.cpp"
    INCLUDE_DIRS "include"
    REQUIRES smooth_component # NOTE: Also tried just `smooth`, `smooth/smooth_component` and those did not work either
)
```

I can use smooth just fine inside of `main` so I am probably doing something foolish if you don't mind pointing me in the right direction.

PerMalmberg, 2021-08-07:
Smooth is meant to be used as a component. Did you write the #include correctly?

jsmestad, 2021-08-07:
Sorry I meant to be used "in a component" meaning I have a regular esp-idf component in `components/` but I cannot have an include (like `#include <smooth/....>`) from inside the esp-idf component.

If I move the same code that was in the esp-idf component into the `main` directory, everything works fine.

PerMalmberg, 2021-08-07:
Oh like a secondary component dependency? I don't know, I've never tried.

jsmestad, 2021-08-07:
Yep that's probably the right term for it :upside_down_face: 

Ah ok, thanks :+1:

closed 2021-08-09 by PerMalmberg